srsran/srsRAN_4G #941: UE cannot find LTE network

08050736, 2022-07-25:
**Issue Description**
> 
when I run sudo srsepc and sudo srsenb, the corresponding network cannot be found using UE. I modified MCC and MNC and dl_earfn but still can't find it.

** Setup **
> 
Ubuntu18.04
USRP B210
smart phone: Samsung A31 and iPhone 8 Plus

**Run srsepc and srsenb **
[image]

[image]

LongRiver-hash, 2022-07-25:
Is SIM information in your config file user_db.csv?

brendan-mcauliffe, 2022-07-26:
Hi, this looks more like an issue for the mailing list. You can subscribe to the list [here](https://lists.srsran.com/mailman/listinfo/srsran-users). 

I suggest you send an email outlining your set-up as you did above, and attach the configs and logs for srsENB & srsEPC. 

You may also want to have a look at [this app note](https://docs.srsran.com/en/latest/app_notes/source/cots_ue/source/index.html) if you have not done so already. 

Closing this issue.

closed 2022-07-26 by brendan-mcauliffe