srsran/srsRAN_4G #943: cannot install srsRAN

chatofking, 2022-07-28:
## Issue Description ##
[Describe the issue in detail]
cannot install srsRAN
## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
wifi ubuntu 22.04 raspberry pi 4b 4g no RF hardware  sudo apt-get install build-essential cmake libfftw3-dev libmbedtls-dev libboost-program-options-dev libconfig++-dev libsctp-dev 
## Expected Behavior ##
[What you expect to happen]
can pass make command 
## Actual Behaviour ##
[What happens instead e.g. error message]
cc1: all warnings being treated as errors
```
make[2]: *** [lib/src/phy/fec/CMakeFiles/srsran_fec.dir/build.make:608: lib/src/phy/fec/CMakeFiles/srsran_fec.dir/turbo/turbodecoder.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:4485: lib/src/phy/fec/CMakeFiles/srsran_fec.dir/all] Error 2
make: *** [Makefile:166: all] Error 2
```
## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
https://docs.srsran.com/en/latest/app_notes/source/zeromq/source/index.html
## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

brendan-mcauliffe, 2022-08-03:
I suggest you try again, make sure all dependencies are installed **and** are up-to-date. Make sure to follow all steps for installing ZMQ and srsRAN carefully, and that you are not getting errors when downloading dependencies. 

Can you be more specific with your set-up, what platform are you trying to install srsRAN on?

chatofking, 2022-08-03:
raspberry pi 4b 
i try on x86 pc work well.

brendan-mcauliffe, 2022-08-03:
Can you run make again, and attach the full console output? To do this, so the following: 

- open the build folder
- run ``make clean``
- run ``make >& make_output.txt``

Then attach that file here.

chatofking, 2022-08-04:
here is my file
[make_output.txt](https://github.com/srsran/srsRAN/files/9256696/make_output.txt)

brendan-mcauliffe, 2022-08-04:
You should try to downgrade the version of GCC being used, I suggest trying version 9.8.

chatofking, 2022-08-04:
E: Unable to locate package gcc-9.8
E: Couldn't find any package by glob 'gcc-9.8'
E: Couldn't find any package by regex 'gcc-9.8'
E: Unable to locate package g++-9.8
E: Couldn't find any package by glob 'g++-9.8'
E: Couldn't find any package by regex 'g++-9.8'

brendan-mcauliffe, 2022-08-04:
Apologies, the above should read GCC version 9.4.

chatofking, 2022-08-04:
when i try gcc --version
-bash: /usr/bin/gcc: No such file or directory
when i try 
sudo apt install gcc-9 g++-9
it show
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
g++-9 is already the newest version (9.4.0-5ubuntu1).
gcc-9 is already the newest version (9.4.0-5ubuntu1).

brendan-mcauliffe, 2022-08-04:
When you run ``gcc --version`` you should get the following (or similar): 

> gcc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0
> Copyright (C) 2019 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Once you have the correct version of GCC, as above, try to re-build srsRAN.

chatofking, 2022-08-05:
can i ask one more question 
is it work correctly ??
99% tests passed, 4 tests failed out of 1527

Label Time Summary:
chest     =  86.41 sec*proc (22 tests)
fec       =   1.36 sec*proc (11 tests)
ldpc      =   0.02 sec*proc (1 test)
lib       = 2397.87 sec*proc (989 tests)
lte       = 2698.51 sec*proc (924 tests)
nr        = 1063.54 sec*proc (359 tests)
pdcp      =   2.60 sec*proc (3 tests)
phch      = 1833.80 sec*proc (467 tests)
phy       = 2747.70 sec*proc (986 tests)
polar     =   0.01 sec*proc (1 test)
rlc       =  33.84 sec*proc (13 tests)
sched     =   4.94 sec*proc (7 tests)
srsenb    =  23.87 sec*proc (10 tests)
srsue     = 362.40 sec*proc (3 tests)
turbo     =   1.33 sec*proc (9 tests)

Total Test time (real) = 4921.10 sec

The following tests FAILED:
        857 - benchmark_radio_multi_rf (Failed)
        887 - network_utils_test (Subprocess aborted)
        1462 - s1ap_test (Failed)
        1470 - ngap_test (Subprocess aborted)
Errors while running CTest
Output from these tests are in: /home/chatofking/srsRAN/build/Testing/Temporary/LastTest.log
Use "--rerun-failed --output-on-failure" to re-run the failed cases verbosely.
make: *** [Makefile:91: test] Error 8

closed 2022-08-05 by chatofking