srsran/srsRAN_4G #945: 5GSA ping 10.10.45.2 failed and srsenb disconnected after ping

troiszhang, 2022-07-31:
## Issue Description ##
srsenb and srsue can be successfully built,but there is a ping problem when I use directions "sudo ip netns exec ue1 ping 10.45.0.1" and "ping 10.45.0.2" ,and also after ping ,terminal of enb printed Disconnecting rnti=0x4601

## Setup Details ##
All configuration is based on srsRAN docs--srsRAN Application Notes--5GSA End to End

## Actual Behaviour ##
srsenb console
ris@ris:~/srsRAN_5GSA/srsRAN/srsenb$ sudo srsenb enb.conf
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins: 
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file enb.conf...

Built in Release mode using commit f2dff0b7a on branch master.

bind(): Address already in use
Failed to bind on address 127.0.0.7, port 2152: Address already in use
Opening 1 channels in RF device=zmq with args=fail_on_disconnect=true,tx_port=tcp://*:2000,rx_port=tcp://localhost:2001,id=enb,base_srate=11.52e6
Supported RF device list: zmq file
CHx base_srate=11.52e6
CHx id=enb
Current sample rate is 1.92 MHz with a base rate of 11.52 MHz (x6 decimation)
CH0 rx_port=tcp://localhost:2001
CH0 tx_port=tcp://*:2000
CH0 fail_on_disconnect=true
NG connection successful

==== eNodeB started ===
Type <t> to view trace
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Setting frequency: DL=1842.5 Mhz, UL=1747.5 MHz for cc_idx=0 nof_prb=52
RACH:  slot=171, cc=0, preamble=0, offset=0, temp_crnti=0x4601
Disconnecting rnti=0x4601.

srsue console
ris@ris:~/srsRAN_5GSA/srsRAN/srsue$ sudo srsue ue.conf
Active RF plugins: libsrsran_rf_zmq.so
Inactive RF plugins: 
Reading configuration file ue.conf...

Built in Release mode using commit f2dff0b7a on branch master.

Opening 1 channels in RF device=zmq with args=tx_port=tcp://*:2001,rx_port=tcp://localhost:2000,id=ue,base_srate=11.52e6
Supported RF device list: zmq file
CHx base_srate=11.52e6
CHx id=ue
Current sample rate is 1.92 MHz with a base rate of 11.52 MHz (x6 decimation)
CH0 rx_port=tcp://localhost:2000
CH0 tx_port=tcp://*:2001
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Waiting PHY to initialize ... done!
Attaching UE...
Random Access Transmission: prach_occasion=0, preamble_index=0, ra-rnti=0xf, tti=171
Random Access Complete.     c-rnti=0x4601, ta=0
RRC Connected
RRC NR reconfiguration successful.
PDU Session Establishment successful. IP: 10.45.0.2
RRC NR reconfiguration successful.

troiszhang, 2022-07-31:
If you are Chinese, you can use Chinese to commment hhhh

brendan-mcauliffe, 2022-08-03:
Hi, can you direct this to the [mailing list](https://lists.srsran.com/mailman/listinfo/srsran-users)? 

Send a mail outlining your issue as you have done here, and also include your logs and config files. 

I will leave this issue open until then.

troiszhang, 2022-08-12:
> Hi, can you direct this to the [mailing list](https://lists.srsran.com/mailman/listinfo/srsran-users)?
> 
> Send a mail outlining your issue as you have done here, and also include your logs and config files.
> 
> I will leave this issue open until then.

ok

closed 2022-11-11 by andrepuschmann