srsran/srsRAN_4G #946: ERROR: internal is not a valid time source. Ethernet connect

egortaran, 2022-08-01:
### Issue Description
When trying to start srsenb using Ettus N210 I get the following error:
[ERROR] [UHD RF] AssertionError: assertion failed:
internal is not a valid time source.
possible values are: [none, external, external].

### Additional Information
Console messages from srsenb:
`$ srsenb /etc/srsran/enb.conf`
```
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_soapy.so libsrsran_rf_zmq.so
Inactive RF plugins: 
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file /etc/srsran/enb.conf...

Built in Release mode using commit f2dff0b7a on branch HEAD.

srsLog error - Unable to create log file "/mnt/srslte/enb.log": No such file or directory
Opening 1 channels in RF device=UHD with args=fail_on_disconnect=true,tx_port=tcp://*:2005,rx_port=tcp://localhost:2008,id=enb,base_srate=23.04e6
Supported RF device list: UHD soapy zmq file
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.2.0.0-0ubuntu1~focal1
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: fail_on_disconnect=true,tx_port=tcp://*:2001,rx_port=tcp://localhost:2002,id=enb,base_srate=23.04e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [USRP2] Opening a USRP2/N-Series device...
[INFO] [USRP2] Current recv frame size: 1472 bytes
[INFO] [USRP2] Current send frame size: 1472 bytes
/home/taran/docker_open5gs/srsRAN/lib/src/phy/rf/rf_uhd_imp.cc:932: uhd_init failed, freeing...
[ERROR] [UHD RF] AssertionError: assertion failed:
  internal is not a valid time source.
  possible values are: [none, external, _external_, mimo].

Error initializing radio.
```
**I use Ethernet to connect. Does srsRAN support my build? Who should be done to avoid this problem?**

andrepuschmann, 2022-08-01:
The N210 is NOT supported by srsRAN. Also not the B100. We've discussed this many times.

closed 2022-08-01 by andrepuschmann

wai-ming-chan, 2022-09-19:
> The N210 is NOT supported by srsRAN. Also not the B100. We've discussed this many times.

I also found out that srsRAN cannot be run in my N210...

Where can I find the list of srsRAN-supporting SDR hardware? In the [official documentation](https://docs.srsran.com/en/latest/app_notes/source/hw_packs/source/index.html), there are only three models (Lime SDR mini 2.0, BladeRF micro 2.0 xA4, and Ettus x310). Is there any other supporting hardware options? 
Thank you!