srsran/srsRAN_4G #947: GNU Radio Broker with Hardware

backmanj, 2022-08-01:
## Issue Description ##
Hello! We are trying to use GNU Radio and srsRAN to create a flow chart where a UE receives and sends its signals through ZMQ source and sink blocks. 

## Setup Details ##
Two Linux Computers that run on Ubuntu 22.04
GNU Radio v 3.9
Two B200mini USRPs
We are doing over the air transmission

## Expected Behavior ##
We want the UE and the eNodeB connect normally with the broker acting in between so we can easily see and modify the IQ values coming into the system.

## Actual Behaviour ##
We have been able to successfully have a UE and eNodeB connect across these two computers and USRPs without GNU radio. When we try to run it with our flow graph, it cannot connect successfully. The UE can find cells, but after it tries to synchronize with random access transmission it never connects.

## Steps to reproduce the problem ##
We have been doing everything on the command file using these commands:
`sudo ./srsRAN/build/srsepc/src/srsepc`
`sudo srsenb`
`sudo .srsRAN/build/srsue/src/srsue --rf.device_name=zmq --rf.device_args="tx_port=tcp://*:2001,rx_port=tcp://localhost:2000,id=ue,base_srate=11.52e6"`

This is our GNU Radio flowgraph 
[image]

## Additional Information ##
 Here is an example of the output on the terminal
`Found Cell:  Mode=FDD, PCI=1, PRB=50, Ports=1, CP=Normal, CFO=-0.8 KHz
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Samp Rate : 11.520000
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Found PLMN:  Id=00101, TAC=7
Random Access Transmission: seq=11, tti=3701, ra-rnti=0x2
Random Access Transmission: seq=13, tti=3721, ra-rnti=0x2
Random Access Transmission: seq=27, tti=3741, ra-rnti=0x2
Random Access Transmission: seq=26, tti=3761, ra-rnti=0x2
Random Access Transmission: seq=49, tti=3781, ra-rnti=0x2
Random Access Transmission: seq=4, tti=3801, ra-rnti=0x2
Random Access Transmission: seq=33, tti=3821, ra-rnti=0x2
Random Access Transmission: seq=17, tti=3841, ra-rnti=0x2
Random Access Transmission: seq=35, tti=3861, ra-rnti=0x2
Random Access Transmission: seq=31, tti=3881, ra-rnti=0x2
[zmq] Error: tx time is 0.136 ms in the past (51497403 < 51498964)
[zmq] Error: tx time is 0.078 ms in the past (103677117 < 103678011)
[zmq] Error: tx time is 0.227 ms in the past (125524338 < 125526956)`

Here is what is found in our log files for the UE
`2022-08-01T21:23:52.722879 [PHY    ] [W] [ 5695] SFN not yet synchronized, sending out-of-sync
2022-08-01T21:23:52.722962 [PHY    ] [W] [ 5695] SNR=1.4 dB RSRP=-46.7 dBm, sync=out-of-sync from channel estimator
2022-08-01T21:23:52.724029 [PHY    ] [W] [ 5696] SFN not yet synchronized, sending out-of-sync
2022-08-01T21:23:52.724169 [PHY    ] [W] [ 5696] SNR=1.5 dB RSRP=-46.7 dBm, sync=out-of-sync from channel estimator
2022-08-01T21:23:52.725131 [PHY    ] [W] [ 5697] SFN not yet synchronized, sending out-of-sync
2022-08-01T21:23:52.725229 [PHY    ] [W] [ 5697] SNR=1.8 dB RSRP=-47.1 dBm, sync=out-of-sync from channel estimator
2022-08-01T21:23:52.726227 [PHY    ] [W] [ 5698] SFN not yet synchronized, sending out-of-sync
2022-08-01T21:23:52.726307 [PHY    ] [W] [ 5698] SNR=1.7 dB RSRP=-47.9 dBm, sync=out-of-sync from channel estimator
2022-08-01T21:23:52.727317 [PHY    ] [W] [ 5699] SFN not yet synchronized, sending out-of-sync
2022-08-01T21:23:52.727408 [PHY    ] [W] [ 5699] SNR=1.5 dB RSRP=-48.6 dBm, sync=out-of-sync from channel estimator`

brendan-mcauliffe, 2022-08-03:
Closing this issue as a response was given on the mailing list.

closed 2022-08-03 by brendan-mcauliffe

FengShaocong, 2023-04-19:
Have you solved the problem？ @backmanj

FengShaocong, 2023-04-19:
Have you solved the problem？I can't find the response on the mailing list. @backmanj

FengShaocong, 2023-04-19:
Have you solved the problem？I can't find the solution on the mailing list. @backmanj

seoyuloh, 2024-12-05:
Hello, how did you solve this problem? could you please share the .grc file of your setup?