srsran/srsRAN_4G #949: UE suddenly disconnected to the network

neilrep, 2022-08-03:
## Issue Description ##
2022-08-03T07:17:38.645033 [RRC    ] [E] Failed to push radio link UL state

## Setup Details ##
We're running an NextEPC v0.3.10--> enodeB --> SDR (b210) --> duplexer --> antenna . A complete set-up for a box base station. SRSENB 21.10 ; UHD 3.15.0

## Expected Behavior ##
All camped UE's stable connected to the network and can stream internet seamlessly

## Actual Behaviour ##
All UE's that connected to the network suddenly disconnected and on the logs shows (2022-08-03T07:17:38.645033 [RRC    ] [E] Failed to push radio link UL state)
## Steps to reproduce the problem ##
Restarting all processes

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

andrepuschmann, 2022-09-21:
Again, provide full logs.

7sAb123swz, 2024-08-28:
Kidding me? It is just a epc network...