srsran/srsRAN_4G #951: questionable eNodeB logs

neilrep, 2022-08-03:
## Issue Description ##
Is that normal logs that you can see while running the enb process and continuously function the UE's

## Setup Details ##
We're running an Commercial EPC--> enodeB --> SDR (b210) --> duplexer --> antenna . A complete set-up for a box base station. SRSENB 21.10 ; UHD 3.15.0

## Expected Behavior ##
All camped UE's stable connected to the network and can stream internet seamlessly

## Actual Behaviour ##
2022-08-03T07:26:53.427346 [PDCP   ] [W] Setting discard timer to 1500ms, to avoid issues with lingering SDUs in the Unacknowledged SDUs map. LCID=3
2022-08-03T07:26:53.427513 [PDCP   ] [W] Setting discard timer to 1500ms, to avoid issues with lingering SDUs in the Unacknowledged SDUs map. LCID=4
2022-08-03T07:26:53.465739 [GTPU   ] [E] Sending SDU with invalid IP version to SPGW
2022-08-03T07:26:53.465740 [GTPU   ] [E] Invalid IP version to SPGW
2022-08-03T07:26:53.585739 [GTPU   ] [E] Sending SDU with invalid IP version to SPGW
2022-08-03T07:26:53.585740 [GTPU   ] [E] Invalid IP version to SPGW
2022-08-03T07:26:53.982217 [RLC    ] [W] SRB1 Signaling max number of reTx=4 for SN=0
2022-08-03T07:26:58.780451 [STCK   ] [E] Bearers: No EPS bearer registered for rnti=0x49
2022-08-03T07:27:05.285539 [S1AP   ] [W] Not handling feature - HandoverRestrictionList
2022-08-03T07:27:05.285541 [S1AP   ] [W] Not handling feature - SRVCCOperationPossible
2022-08-03T07:27:05.339096 [PDCP   ] [W] Setting discard timer to 1500ms, to avoid issues with lingering SDUs in the Unacknowledged SDUs map. LCID=3
2022-08-03T07:27:05.909244 [PDCP   ] [W] Setting discard timer to 1500ms, to avoid issues with lingering SDUs in the Unacknowledged SDUs map. LCID=3
2022-08-03T07:27:06.704835 [PDCP   ] [W] Setting discard timer to 1500ms, to avoid issues with lingering SDUs in the Unacknowledged SDUs map. LCID=4
2022-08-03T07:27:09.451681 [RLC    ] [W] SRB1 Signaling max number of reTx=4 for SN=1
2022-08-03T07:27:14.322587 [STCK   ] [E] Bearers: No EPS bearer registered for rnti=0x4c
2022-08-03T07:27:26.274844 [S1AP   ] [W] Not handling feature - HandoverRestrictionList
2022-08-03T07:27:26.274845 [S1AP   ] [W] Not handling feature - SRVCCOUser 0x4e connected

All UE's connected to the network and can stream internet via Youtube. Just asking a question. As the logs that you can see above, is that normal while all UE's connected to the network and streaming YouTube? We have no idea about this appearing logs. And if you encountered it the same, how can you handle it? 

## Steps to reproduce the problem ##
Continuing the process as the function continuously running

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

zikkuratti, 2022-09-19:
Which bandwidth been used? Saw same warnings today on 1,4Mhz.

andrepuschmann, 2022-09-21:
From the eNB logs it seems that the core is using some messages that the eNB's S1AP doesn't support so it's logging that. Consider turning the feature of in the core to avoid those logs.

Further the eNB logs:
```
2022-08-03T07:26:53.585739 [GTPU ] [E] Sending SDU with invalid IP version to SPGW
2022-08-03T07:26:53.585740 [GTPU ] [E] Invalid IP version to SPGW
```
which indicates something weird (neither IPv4 nor IPv6) was received over PDCP. You would need to check your Wireshark to see what being sent there.

I am closing the issue. Please reopen if there is a more specific question.
Thanks

closed 2022-09-21 by andrepuschmann