srsran/srsRAN_4G #952: E: Unable to locate package srsepc

chatofking, 2022-08-05:
## Issue Description ##
[Describe the issue in detail]
I can't install srsEPC
## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
ubuntu 22.04 raspberry pi 4b 4g zeromq 
## Expected Behavior ##
[What you expect to happen]
can install srsEPC

## Actual Behaviour ##
[What happens instead e.g. error message]
```
$ sudo add-apt-repository ppa:srslte/releases
Repository: 'deb https://ppa.launchpadcontent.net/srslte/releases/ubuntu/ jammy main'
Description:
This is the Ubuntu PPA for srsLTE, a free and open-source LTE software suite, along with some dependencies.

For more info, please visit https://github.com/srsLTE/srsLTE
More info: https://launchpad.net/~srslte/+archive/ubuntu/releases
Adding repository.
Press [ENTER] to continue or Ctrl-c to cancel.
Found existing deb entry in /etc/apt/sources.list.d/srslte-ubuntu-releases-jammy.list
Adding deb entry to /etc/apt/sources.list.d/srslte-ubuntu-releases-jammy.list
Found existing deb-src entry in /etc/apt/sources.list.d/srslte-ubuntu-releases-jammy.list
Adding disabled deb-src entry to /etc/apt/sources.list.d/srslte-ubuntu-releases-jammy.list
Adding key to /etc/apt/trusted.gpg.d/srslte-ubuntu-releases.gpg with fingerprint 2C68ED1E6E219E244A6629BB3BD852360A6C26DE
Hit:1 http://ports.ubuntu.com/ubuntu-ports jammy InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease
Hit:4 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease
Ign:5 https://ppa.launchpadcontent.net/srslte/releases/ubuntu jammy InRelease
Err:6 https://ppa.launchpadcontent.net/srslte/releases/ubuntu jammy Release
  404  Not Found [IP: 2620:2d:4000:1::3e 443]
Reading package lists... Done
E: The repository 'https://ppa.launchpadcontent.net/srslte/releases/ubuntu jammy Release' does not have a Release file.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

`$ sudo apt-get install srsepc
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package srsepc
`

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

zikkuratti, 2022-08-08:
Changing srsLTE to srsRAN can solve this  problem, i think

egortaran, 2022-08-09:
If you use Ubuntu 22.04, then you need to go to "programs and updates" -> "Other software". Remove the checkboxes from Jammy. After that, repeat the step "$ sudo add-apt-repository ppa:srslte/releases". It helped me

brendan-mcauliffe, 2022-08-09:
As mentioned above, you should be using the srsRAN packages as these are the most up-to-date. 

``sudo add-apt-repository ppa:softwareradiosystems/srsran``
``sudo apt-get update``
``sudo apt-get install srsran -y``

chatofking, 2022-08-12:
> As mentioned above, you should be using the srsRAN packages as these are the most up-to-date.
> 
> `sudo add-apt-repository ppa:softwareradiosystems/srsran` `sudo apt-get update` `sudo apt-get install srsran -y`
> 
> First add the package repository, then you can easily install each of the individual applications. Just swap out the last line, eg: `sudo apt-get install srsepc` instead of `sudo apt-get install srsran`.

This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'srsepc' has no installation candidate

brendan-mcauliffe, 2022-08-12:
Run ``sudo apt-get update`` again, and make sure there are no issues or errors. I can run the above locally and successfully install srsepc.

chatofking, 2022-08-12:
$ sudo apt-get update
Hit:1 http://ports.ubuntu.com/ubuntu-ports jammy InRelease
Hit:2 http://ports.ubuntu.com/ubuntu-ports jammy-updates InRelease
Hit:3 http://ports.ubuntu.com/ubuntu-ports jammy-backports InRelease
Hit:4 https://ppa.launchpadcontent.net/softwareradiosystems/srsran/ubuntu jammy InRelease
Hit:5 http://ports.ubuntu.com/ubuntu-ports jammy-security InRelease
Reading package lists... Done
i still got same error
```
Package srsepc is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'srsepc' has no installation candidate
```

chatofking, 2022-08-13:
> If you use Ubuntu 22.04, then you need to go to "programs and updates" -> "Other software". Remove the checkboxes from Jammy. After that, repeat the step "$ sudo add-apt-repository ppa:srslte/releases". It helped me
I use ubuntu server which command is it ??

closed 2022-08-22 by chatofking