srsran/srsRAN_4G #955: different MCC MNC on configures and slice not support

hyunseoha, 2022-08-10:
Hello,
i am on the construction of private 5g network.

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
I am using OPEN5GS, srsENB and ZMQ currently. The private network without any radio hardware.

## Expected Behavior ##
i wanted to used MCC MNC different from the examples from the following link,
https://docs.srsran.com/en/latest/app_notes/source/5g_sa_E2E/source/index.html#enb .

with the configuration examples it worked perfectly.

## Steps to reproduce the problem ##
however, once i changed the following two values from the configuration file enb.conf and amf.yaml 

MCC 001
MNC 01

## Actual Behaviour ##
it is not working and the complain is as follow.

Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_zmq.so
Inactive RF plugins: 
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file enb.conf...

Built in Release mode using commit f2dff0b7a on branch master.

Opening 1 channels in RF device=zmq with args=fail_on_disconnect=true,tx_port=tcp://*:2000,rx_port=tcp://localhost:2001,id=enb,base_srate=11.52e6
Supported RF device list: UHD zmq file
CHx base_srate=11.52e6
CHx id=enb
Current sample rate is 1.92 MHz with a base rate of 11.52 MHz (x6 decimation)
CH0 rx_port=tcp://localhost:2001
CH0 tx_port=tcp://*:2000
CH0 fail_on_disconnect=true
_**NG Setup Failure. Cause: radioNetwork - slice-not-supported**_

==== eNodeB started ===
Type <t> to view trace
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Current sample rate is 11.52 MHz with a base rate of 11.52 MHz (x1 decimation)
Setting frequency: DL=1842.5 Mhz, UL=1747.5 MHz for cc_idx=0 nof_prb=52

thanks for helping me !

Leo

zikkuratti, 2022-08-11:
not shure about hardware & 5G but nof_prb=52 confuse me, you can try 25, 50, 75 etc.

hyunseoha, 2022-08-11:
Thanks for the answer.

Do you mean that i have to change the value of nof_pro from 52 to 25 ?

Can you tell me what the value means and what i expect by changing it ?

Thanks

Leo

zikkuratti, 2022-08-11:
nof_prb means bandwidth and it hardcoded for band 3 u try to use it-s 6,15,25,50,75,100

hhyunseo, 2022-08-16:
Thanks for the answer again.

I used all the configuration files from the website here.
https://docs.srsran.com/en/latest/app_notes/source/5g_sa_E2E/source/index.html#

then the change of MCC MNC worked, i don't know, but anyway worked.

@zikkuratti i still don't get your advice. i changed the line BAND from the file rr.conf from 3 to something else.

rr.conf as following
nr_cell_list =
(
  {
    rf_port = 0;
    cell_id = 1;
    root_seq_idx = 1;
    tac = 7;
    pci = 500;
    dl_arfcn = 368500;
    band = 3;	<----------- this number to another
  }
);

is it what you said ?
if not can you give me the exact file and line please ?
i want to try changing nof_prb value more than ever, lol.

Thanks,

Leo

andrepuschmann, 2022-09-21:
The nof_prb 52 is correct for 10 MHz NR, for LTE it was 50. If the core says it doesn't support the "slice" maybe you don't have the PLMN configured correctly. Anyway, since you said it's working now I am going to close the issue.

closed 2022-09-21 by andrepuschmann