srsran/srsRAN_4G #956: UE won't connect with zmq loopback

kmski, 2022-08-12:
## Issue Description ##
The UE says it's Attaching but never completes.  There is no "Cell Search" shown on the UE console as the attach simply hangs.

## Setup Details ##
Trying to setup 5G SA using Open5GS, srsenb and srsue in a ZeroMQ loopback.

## Expected Behavior ##
The srsenb connects to the Open5GS core, the srsue registers with the Open5GS core via the srsenb and completes the loopback connection.

## Actual Behaviour ##
The Core and gNB connect OK.  The UE tries attaching but hangs and continually prints "Disconnected" on the trace output.
[enb.conf.txt](https://github.com/srsran/srsRAN/files/9322752/enb.conf.txt)
[enb.log](https://github.com/srsran/srsRAN/files/9322794/enb.log)
[rr.conf.txt](https://github.com/srsran/srsRAN/files/9322993/rr.conf.txt)
[ue.conf.txt](https://github.com/srsran/srsRAN/files/9322997/ue.conf.txt)
[ue.log](https://github.com/srsran/srsRAN/files/9323013/ue.log)

## Steps to reproduce the problem ##
Install and start Open5GS with the following yaml configs (note: UE SIM details were entered via Open5GS WebGUI):
[upf.yaml.txt](https://github.com/srsran/srsRAN/files/9323693/upf.yaml.txt)
[amf.yaml.txt](https://github.com/srsran/srsRAN/files/9323706/amf.yaml.txt)

Added network namespace for ue1:
ip netns add ue1

Start the gNB:
srsenb ue.conf

Start the UE:
srsue ue.conf

## Additional Information ##
For some reason, it appears that the Cell connection is failing but I'm at a loss as to why (since this is simply a loopback example).

andrepuschmann, 2022-09-21:
Can you share a full UE log in info along with UE console output please?

kmski, 2022-09-21:
Hello Andre,
Thanks for responding.  I actually got this to work eventually.  As it turns out, I had an issue with sampling rates vs. PRBs.  I was not aware of the secret "lte_sample_rates" expert option.  Although the LTE spec states that a sample rate of 15.36 Mhz should require 50 resource blocks (PRBs), it seems the srsRAN defaults to non-standard rates, so the 15.36 Mhz rate was requiring 75 PRBs, which seems wrong to me.  Anyway, once I straightened out the sample rate issue, things seemed to work much better.