srsran/srsRAN_4G #957: Increasing internet speed

egortaran, 2022-08-15:
# Question
Hello dear developers and users. I'm doing a school radio project. I would like to increase the speed of the mobile Internet. The Host's speedtest shows the speed of 95 Mbps downlink, 97 Mbps uplink. When connecting the phone to srsRAN, the speed of the connection reaches up to 13 Mbps.

## Technology stack
I use the following technology stack:
- ubuntu 22.04
- Open5GS
- kamailio
- docker-compose
- limesdr v1.4s (LimeSuite v20.10.0, SoapySDR 0.7.2)
- two One Plus 9

### enb.conf
- dl_earfcn = 2525
- tx_gain = 80
- rx_gain = 40
- n_prb = 50

## Possible solution
I would like to achieve a speed of 20mbit s. Which parameters in the configs are the best way to change to improve Internet speed.

matan1008, 2022-08-15:
I am not an expert, but in my humble opinion you should try a different band (3 or 7) and you can try to increase the n_prb

NeilJohnR, 2022-08-16:
You should try to band scan your place and try to get the center frequency after getting it, put it on dl_earfcn then lower your bandwidth to 25Mhz = n_prb = 25

egortaran, 2022-08-17:
Thank you very much for your responsiveness! I tried 7 band with links of different bandwidth. I couldn't increase the speed. I changed it to 8 band, took the parameter 25 nprb. Unfortunately the speed has only gotten worse. The speed began to reach 10 Mbps

closed 2022-08-19 by egortaran