srsran/srsRAN_4G #958: Multiple MME or MME Pooling

neilrep, 2022-08-16:
## Issue Description ##
MME Pooling

## Setup Details ##
We're running an Commercial EPC--> enodeB --> SDR (b210) --> duplexer --> antenna . A complete set-up for a box base station. SRSENB 21.10 ; UHD 3.15.0

##Question##
Can we able to point enodeb (srsneb) to multiple mme or mme pooling? in order to try or test the scenarios for voicecalls and sms, if it is working on srs, what is our adjustment we need to do?

zikkuratti, 2022-08-16:
not shure, but try too add renamed duplicates needed to you, change ip's, rename modules and correct add them to kind a docker-compose.yaml on different carefully managed port's i dont think srsRAN give a F  who ask metadatazzz)

andrepuschmann, 2022-09-21:
@neilrep We've never tried MME pooling I am afraid. From the eNB we don't support multiple MME connections with load-balancing or failover if that is what you mean. Maybe there are third-party load-balancer available that could help here.

closed 2022-09-21 by andrepuschmann