srsran/srsRAN_4G #962: srsUE is showing  Failed to setup/configure GW interface  nTp)

prasenjitpdas, 2022-08-17:
## Issue Description ##
the srsUE is showing this error after IP assigment , in [srsRAN with ZMQ Virtual Radios]
Network attach successful. IP: 10.45.0.3
Failed to setup/configure GW interface
 nTp) 16/8/2022 12:50:58 TZ:34

## Setup Details ##
srsUE  --> srsRAN-->5g core , single VM in ubuntu22.x

## Expected Behavior ##
UE is able to ping and vice-versa

## Actual Behaviour ##
not able to ping UE / Core 
## Steps to reproduce the problem ##
followed the  setup as https://docs.srsran.com/en/latest/app_notes/source/zeromq/source/index.html#zeromq-installation

## Additional Information ##
s1AP /PFCP all are success

andrepuschmann, 2022-09-21:
You need to run with `sudo` or pass the correct caps to the `srsue` binary, see https://github.com/srsran/srsRAN/issues/981 for reference.

closed 2022-09-21 by andrepuschmann

Azukaa, 2023-04-01:
I dont understand your solution to this proble. can you please elaborate. I am facing it too

rkalbasi-ipk, 2024-12-31:
I am facing this error as well, Failed to setup/configure GW interface
Using sudo does not help, any solution?