srsran/srsRAN_4G #968: SCTP abort message [S1AP failed to initiate] using open5gs and srsRAN

Citrayaf, 2022-08-23:
## Issue Description ##
I tried to use srsRAN and open5gs for 5G NSA simulation for UE and eNodeB I use srsRAN and for EPC I use open5gs in different VM. when I try to integrate these, from eNb to open5gs always send SCTP abort message if I trace using wireshark. I don't know what is the main problem come from srsRAN or open5gs. 

## Setup Details ##
I use virtual ZeroMQ, epc uses open5gs, for 5G NSA simulation, in different VM
10.14.2.5 is IP for open5gs
10.14.2.7 is IP for srsRAN

## Expected Behavior ##
S1AP procedure is OK

## Actual Behaviour ##
when I try to integrate these, from eNb to open5gs always send SCTP abort message if I trace using wireshark. in enb.log shows that S1AP failed to initiate.
open5Gs
[image]
srsRAN
[image]
## Steps to reproduce the problem ##
uses different VM, epc uses open5gs
## Additional Information ##
I have attached all trace files:
srsRAN 
[enb.log](https://github.com/open5gs/open5gs/files/9399024/enb.log)
[srsRAN_trace.zip](https://github.com/open5gs/open5gs/files/9399028/srsRAN_trace.zip)
open5gs
[open5G_trace.zip](https://github.com/open5gs/open5gs/files/9399029/open5G_trace.zip)

Citrayaf, 2022-08-28:
I have fixed the problems, I misconfigured in srsRAN enb side IP configuration after I change it to the correct one all is working fine. 
Thanks you

closed 2022-08-28 by Citrayaf

yopie176, 2022-08-30:
can you share config amf.yaml, upf.yaml, srsenb/enb.conf and srsenb/rr.conf
thanks

Citrayaf, 2022-09-06:
Sorry for my late reply, sorry right now I can share it to those files to you because I have lot work to do. But you can access my repo in this link https://github.com/Citrayaf/How-to-build-OpenCore-and-OpenRAN-for-5G . I have made documentation about what I have done to build this system, Maybe you can check. 
Thanks very much. Later I will send you those files..