srsran/srsRAN_4G #971: SCS & TDD feature

liyuecheng, 2022-08-29:
## Issue Description ##
Hello,I want to use srsRAN as a baseline to build a 5G nr system at mm-wave band (FR2).  
I have read the documents ( Application notes: 5G SA End-to-End , 5G SA COTS UE, 5G SA srsUE),
and noticed that "Limited to 15 kHz Sub-Carrier Spacing (SCS), which means only FDD bands can be used." mentioned in "5G SA COTS UE" and" 5G SA srsUE", but  not  "5G SA End-to-End". And band 78 is used in "5G SA End-to-End" which supports TDD.  
so I'm confused that is srsRAN supports TDD or not? Is SCS other than 15khz supported? If not, Any plans for those currently limited features?

andrepuschmann, 2022-09-21:
Thanks for the question. I confirm that we do support TDD. That is what's being used in a few NSA configs for example. However, we do not support 30kHz SCS (it's more a limitation of the srsRAN radio layer rather than the signal processing). The reason why SA is only FDD is because practically no commercial phone supports 15kHz on an TDD band. 

We are currently working on a new gNB stack that we are planning to release towards the end of the year - it will support all those configs out-of-the-box.

closed 2022-09-21 by andrepuschmann

hahat1999, 2024-02-24:
Hello 
is there any update to fix this issue?