srsran/srsRAN_4G #974: SA srsUE can't attach commercial network with B210

fengchao-1998, 2022-08-30:
## Issue Description ##
SA srsUE can't attach commercial network with B210;

2022-08-30T01:48:58.142593 [RRC    ] [I] using srand seed of 142593
2022-08-30T01:48:58.437978 [RRC    ] [D] State IDLE
2022-08-30T01:48:58.438006 [NAS5G  ] [D] Initiating RRC NR Connection
2022-08-30T01:48:58.438007 [RRC-NR ] [I] Proc "Setup Request" - Initiation of Setup request procedure
2022-08-30T01:48:58.438008 [RRC-NR ] [I] Proc "Cell Selection" - Starting...
2022-08-30T01:48:58.714694 [RRC-NR ] [I] Proc "Cell Selection" - Completed with failure.
2022-08-30T01:48:58.714699 [RRC-NR ] [W] Could not finish setup request. Deallocating dedicatedInfoNAS PDU

SA srsUE cannot complete cell search

## Setup Details ##
srsRAN  22.04.1

USRP B210

UHD_3.15.0.0-0

Commercial base station configuration：
15 kHz Sub-Carrier Spacing (SCS)
10 MHz Bandwidth (BW)
band 28
arfcn 154090

ue.conf     [ue.txt](https://github.com/srsran/srsRAN/files/9449150/ue.txt)

## Expected Behavior ##
SA srsUE  attach commercial network

## Actual Behaviour ##
SA srsUE cannot complete cell search

## Steps to reproduce the problem ##

## Additional Information ##

[ue.log](https://github.com/srsran/srsRAN/files/9449178/ue.log)

I would like to know which configuration of commercial network SA srsUE supports.
Thank you.

mariatsampazi, 2022-09-14:
Have you found a solution yet?

fengchao-1998, 2022-09-15:
> 你找到解决方案了吗？

Unfortunately it still doesn't work. But I can use srsUE to access 4G commercial network.

fengchao-1998, 2022-09-19:
Modify the configuration in the srsRAN/srsue/src/main.cc directory：

    ("rat.nr.bands",        bpo::value<string>(&args->stack.rrc_nr.supported_bands_nr_str)->default_value("3"),   "Supported NR bands")
    ("rat.nr.nof_carriers", bpo::value<uint32_t>(&args->phy.nof_nr_carriers)->default_value(0),                   "Number of NR carriers")
    ("rat.nr.max_nof_prb",  bpo::value<uint32_t>(&args->phy.nr_max_nof_prb)->default_value(52),                   "Maximum NR carrier bandwidth in PRB")
    ("rat.nr.dl_nr_arfcn",  bpo::value<uint32_t>(&args->stack.rrc_nr.dl_nr_arfcn)->default_value(368500),         "DL ARFCN of NR cell")
    ("rat.nr.ssb_nr_arfcn", bpo::value<uint32_t>(&args->stack.rrc_nr.ssb_nr_arfcn)->default_value(368410),        "SSB ARFCN of NR cell")
    ("rat.nr.nof_prb",      bpo::value<uint32_t>(&args->stack.rrc_nr.nof_prb)->default_value(52),                 "Actual NR carrier bandwidth in PRB")
    ("rat.nr.scs",          bpo::value<string>(&scs_khz)->default_value("15"),                                    "PDSCH subcarrier spacing in kHz")
    ("rat.nr.ssb_scs",      bpo::value<string>(&ssb_scs_khz)->default_value("15"),                                "SSB subcarrier spacing in kHz")

I modified the band, ssb arfcn, DL arfcn in the configuration

closed 2022-09-19 by fengchao-1998