srsran/srsRAN_4G #975: 5G-SA:  cell configuration for Band n78:  ERROR: Coreset0 index=6 is not compatible with DL ARFCN 630000 in band 78

chris5g, 2022-08-31:
## Issue Description ##

I am not able to configure Band n78 with srsRAN v220401 in rr.conf for 5G SA.
Other bands (e.g. n3, n71) are ok (at least srsenb is starting).  
I can't figure out what coreset0 index is working.....for Band n78. 
When starting srsenb I get the following ERROR:

ERROR: Coreset0 index=6 is not compatible with DL ARFCN 630000 in band 78
ERROR: Deriving SSB parameters

## Setup Details ##
nr_cell_list =
(
  // no NR cells
  {
    rf_port = 0;
    cell_id = 1;
    root_seq_idx = 1;
    tac = 7;
    pci = 500;

/*
    // FDD:
    dl_arfcn = 368500;
    coreset0_idx = 6
    band = 3;
*/

    // TDD:
    dl_arfcn = 630000;
    band = 78;
    coreset0_idx = 6;

  }
);

Any help would be appriciated.

strongcourage, 2022-09-05:
Hi @chris5g, I tried also to configure Band n78 for the NR cell as my Samsung S20 worked with 5G NSA network. However, for 5G SA network, I faced the same issue as yours. I'm wondering whether you've figured out the solution? Or we really need to buy a new phone in the list of [tested phones](https://docs.srsran.com/en/latest/app_notes/source/5g_sa_COTS/source/index.html#tested-devices). Thanks.

chris5g, 2022-09-05:
Hi,
unfortunately now. I found some note in the srsRAN docu:  
"Currently, srsRAN only supports the use of FDD bands for 5G SA." Since n78 is TDD I fear that n78 is currently not supported. any confirmation would be appriciated. 

btw. I have a Samsung S21 and SA is not working for me at currently. ( I guess firmware update needed). however at least the 5G signal can be seen.

hope that helps,
br,
chris

strongcourage, 2022-09-05:
Thanks @chris5g . How did you check this "at least the 5G signal can be seen"?

chris5g, 2022-09-05:
I searched for network operators manually in the mobile network menu of the phone and I could saw the MCC,NDC from the 5GSA.

chris5g, 2022-09-05:
Just to let know and not confuse u. The Samsung S21 was used with another radio not with srsRAN when I mentioned to see the 5g SA signal.

iman432, 2022-09-06:
i also have this issue even by FDD , for example in my case band is N28 :

[image]

chris5g, 2022-09-06:
@iman432:  With this setting gnB is starting:
     dl_arfcn = 156000;
     band = 28;
     coreset0_idx = 6;

strongcourage, 2022-09-06:
@chris5g How did you know the correct value of `dl_arfcn`? Is there any official document about this parameter for each band? Also, I've tried to modify the value of `coreset0_idx` for band 78, but had no luck.

iman432, 2022-09-06:
> @iman432: With this setting gnB is starting: dl_arfcn = 156000; band = 28; coreset0_idx = 6;

Thank you, my version is 22.04 and i think its solved in 22.04.1 as below :

Change Log for Releases
=======================

## 22.04.1
  * Various bug fixes in RLC AM and PDCP for NR
  * Fix crash when UE attempted to reestablish in SA
  * Remove fixed coreset0 index for SSB ===  should be related to this
  * Add support for SIB5 and SIB6 transmission in LTE

chris5g, 2022-09-06:
@iman4332:  Yes I using v22.04.01.

 @strongcourage:  there a several sites available. e.g. https://www.sqimway.com/nr_band.php/

iman432, 2022-09-19:
Seems bands after N71, have issue with coreset.

N78 :

[image]

N77 :

[image]

N74 :

[image]

robertfalkenberg, 2022-09-20:
Currently for 5G-SA, TDD bands with 30kHz SCS (e.g. n78, n77) are not supported.

closed 2022-09-20 by robertfalkenberg

BouazzaEhm, 2024-03-27:
Hello  i have the same issue using 5G NSA can someone help me please

BouazzaEhm, 2024-03-27:
nr_cell_list =
(
   {
    rf_port = 1;
    cell_id = 0x02;
    tac = 0x0007;
    pci = 500;
    root_seq_idx = 204;

    // TDD:
    //dl_arfcn = 634240;
    //band = 78;

    // FDD:
    dl_arfcn = 368500;
    band = 3;
    coreset0_idx = 6

  }
);

kata1219, 2025-03-28:
@BouazzaEhm Maybe try srsRAN. I’ve tested srsRAN 5G NSA with N310 and it can successfully connect to a COTS UE.