srsran/srsRAN_4G #976: N3 is missing GTP-U PDU session container extension header

mk122112, 2022-08-31:
## Issue Description ##

N3 is missing GTP-U PDU session container extension header.
It looks more like an S1-U interface.

## Setup Details ##

srsENB, srsUE and Core in separate VMs.

srsran release 22.04.1

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

mk122112, 2022-09-01:
[image]
[image]

andrepuschmann, 2022-09-21:
@mk122112 Thanks for reporting this. I confirm that we don't pack the PDU session container. We simply didn't need it for our purposes. There are currently also no plans to add it anytime soon. That being said, PRs are of course open. Closing this.

closed 2022-09-21 by andrepuschmann

mk122112, 2022-09-21:
I was using srsRAN with a different 5GC and user plane is not working due to this. But I don't need it right now.

andrepuschmann, 2022-09-21:
> I was using srsRAN with a different 5GC and user plane is not working due to this. But I don't need it right now.

Ah, interesting. We assumed it wasn't mandatory. Can you disclose the particular 5GC you've used for this?

mk122112, 2022-09-21:
It's simulated 5GC developed by Ericsson for test.