srsran/srsRAN_4G #978: srsenb execute error

beruysaliz, 2022-09-01:
## Issue Description ##
when I execute srsenb taking runtimeerror. I attached enb.log.
[enb.log](https://github.com/srsran/srsRAN/files/9470907/enb.log)
[image]

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
USRP N321
srsenb config file attached.
[srsenb.txt](https://github.com/srsran/srsRAN/files/9470945/srsenb.txt)

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

robertfalkenberg, 2022-09-20:
According to [this overview](https://kb.ettus.com/N320/N321) the USRP N321 only supports the following sampling rates: 200, 245.76, 250 MS/s.

srsRAN requires the following sampling rates for LTE:
* 25PRB: 5.76MS/s or 7.68MS/s
* 50PRB: 11.52MS/s or 15.36MS/s
* 75PRB: 15.36MS/s or 23.04MS/s
* 100PRB: 23.04MS/s or 30.72MS/s

closed 2022-09-20 by robertfalkenberg

andrepuschmann, 2022-09-20:
To follow up on what Robert said, you can use the N310 instructions and run with LTE rates. See [here](https://docs.srsran.com/en/latest/app_notes/source/5g_nsa_cots/source/index.html#ettus-research-usrp-n310). For 50 PRB this will use 15.36msamps/s with 245.76/15.36=16 decimation.

beruysaliz, 2022-09-21:
> 

Thank you for your reply robert. Iset master clock 250 MS/s, but I couldn't set decimation rate to a value that I want.
here [https://kb.ettus.com/USRP_N300/N310/N320/N321_Getting_Started_Guide#N320.2FN321_Example_Sample_Rates](url) in that link there are sample rates. My sample rate is always 11MS/s.

beruysaliz, 2022-09-21:
> 

Thank you for reply andrea. How can I set PRB value for my N321 ?

robertfalkenberg, 2022-09-22:
> Thank you for your reply robert. Iset master clock 250 MS/s, but I couldn't set decimation rate to a value that I want.
> here [...]

As @andrepuschmann suggested, please try to set `master_clock_rate=245.76e6` in the `device_args`

robertfalkenberg, 2022-09-22:
Also remember to compile with LTE rates enabled, i.e. `cmake .... -DUSE_LTE_RATES=True ....`, for 25, 50, and 100 PRB cell.