srsran/srsRAN_4G #981: 5G SA End to End using Open5GS and srsRAN with ZMQ -  Failed to setup/configure GW interface when start srsue

Edgard21031969, 2022-09-05:
## Issue Description ##
[I used the link https://docs.srsran.com/en/hoverxref_test/app_notes/source/5g_sa_E2E/source/index.html to simulate the 5G SA network.]

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

## Expected Behavior ##
[The connection should be successfully established once the UE has been assigned an IP address]

## Actual Behaviour ##
[When I launch srsenb and srsue, they started. But after, I have on ue the following error : Failed to setup/configure GW interface
RRC Connected
Failed to setup/configure GW interface
RRC NR reconfiguration successful
Received RRC Release
]

## Steps to reproduce the problem ##
[I reproduce the same configuration as showing on the link: ue.conf, rr.conf, enb.conf, amf.yaml, upf.yaml] 
[image]
[image]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

robertfalkenberg, 2022-09-19:
The UE needs permission to spawn a `tun` network device. The easiest way is to run the application as `root` via `sudo`.

robertfalkenberg, 2022-09-19:
Another option is to configure the capabilities for the `srsue` binary

```
sudo setcap cap_net_admin+eip <...>/build/srsue/src/srsue
```
after each build.

andrepuschmann, 2022-09-21:
Closing this after @robertfalkenberg provided the solution.

closed 2022-09-21 by andrepuschmann

chaohwa4, 2023-03-24:
After running as a root or using sudo, I still have the Failed to setup/configure GW interface issue from the UE hosted by a Ubutu 20.04 laptop.

stadekawa, 2023-04-01:
Make sure you created the network namespace.
`sudo ip netns add ue1`

walidouaali, 2025-09-01:
a common issue is caused by not isolating the UE network. To fix this, simply create a network namespace for the UE by running: 
sudo ip netns add ue1