srsran/srsRAN_4G #982: can anyone help me in implemenating srsRAN on Raspberry pi 4

anum63, 2022-09-06:
## Issue Description ##
[Describe the issue in detail]

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

TallGuy74, 2022-09-15:
@anum63 I have a 4G UE/eNB running on Odroid N2+ hardware using USRP X300's connected to the internal ethernet controller. You can probably get this setup working on the Raspberry Pi 4 as well, but it's possible the CPU can't be run fast enough. You WILL need to overclock the CPU to at least 2 GHz.

robertfalkenberg, 2022-09-19:
Please have a look here: https://docs.srsran.com/en/latest/app_notes/source/pi4/source/index.html

andrepuschmann, 2022-09-21:
I am closing the issue. Sufficient information has been provided in the appnote and this ticket. Please reopen should more specific questions arise.

closed 2022-09-21 by andrepuschmann