srsran/srsRAN_4G #984: No cell_search_nbiot in build/lib/examples

frycaktadeas, 2022-09-08:
## Issue Description ##
Hello,

I want to run `cell_search_nbiot`, but I did not find it anywhere after installing srsran from APT, so I have installed it from source by this [manual](https://docs.srslte.com/en/latest/general/source/1_installation.html)
But after build no `cell_search_nbiot` was compiled in `build/lib/examples`

## Setup Details ##
Ubuntu 22.04

## Expected Behavior ##
I have expected to find `cell_search_nbiot` in `build/lib/examples` like [this page](https://docs.srslte.com/en/latest/app_notes/source/nbiot/source/) says

## Steps to reproduce the problem ##
Run all commands from this [manual](https://docs.srslte.com/en/latest/general/source/1_installation.html

robertfalkenberg, 2022-09-19:
Please install at least one RF driver: https://docs.srslte.com/en/latest/general/source/1_installation.html#installation-from-source
Otherwise all RF-related binaries won't be built.

frycaktadeas, 2022-09-19:
Of course I have had installed BladeRF driver before compile.

robertfalkenberg, 2022-09-19:
Could you please provide the console output of the cmake and build run in a *clean* build directory?

andrepuschmann, 2022-11-07:
Closing this due to inactivity. I've also double checked on my local built of 22.10 and it does compile the `cell_search_nbiot` in examples.