srsran/srsRAN_4G #985: 5G SA - Cannot connect srsUE to srsENB using USRP X310

nathanstep55, 2022-09-08:
## Issue Description ##
Hello,
I have been trying to connect srsUE to srsENB between two computers using two USRP X310s. Despite confirming that the USRPs can see the signals using the UHD test waveform and RX DFT examples, I have not been successful in getting the srsRAN software to communicate.

## Setup Details ##
Two Ubuntu host PCs running Ubuntu 20.04 inside VirtualBox with bridged adapters so that they act as separate devices on the network. These PCs are not connected to any network besides their connection with the USRP RF hardware. srsRAN 22.04.1 is being used with a USRP X310 on each computer containing a single UBX-160 daughterboard each with two antennas attached to each daughterboard. UHD 4.2.0 is being used on both computers and the 1Gbit interface is being used, not 10Gbit.

## Expected Behavior ##
The UE connects to the base station and I am given an IP address for the UE. I should be able to ping the computers from both sides.

## Actual Behavior ##
The UE tries to find a cell and attach to the base station, but fails to do so. Using the `rx_ascii_art_dft` script shows that the USRP X310 attached to the UE is able to receive the signal at ~1800MHz (NR band 3) from srsENB. Sometimes the RF times out, but regardless of whether it does or not, the same behavior occurs.

## Steps to reproduce the problem ##
Follow the instructions [here](https://docs.srslte.com/en/latest/app_notes/source/5g_sa_E2E/source/index.html) but instead of using ZeroMQ, change the configuration to use UHD driver with the following device arguments:

`device_args = type=x300,clock=internal,sampling_rate=11.52e6,lo_freq_offset_hz=23.04e6,send_frame_size=1472,recv_frame_size=1472,num_send_frames=64,num_recv_frames=64,None`

## Additional Information ##
The logs from the srsENB and srsUE commands are attached. We have also tried using srsRAN’s `cell_search` tool but it freezes after the first 50 iterations or so. srsENB shows RF underflow and srsUE shows RF overflow, if that is relevant.

[srsgnb3.log](https://github.com/srsran/srsRAN/files/9530706/srsgnb3.log)
[srsue3.log](https://github.com/srsran/srsRAN/files/9530709/srsue3.log)

gabri94, 2022-09-13:
It seems that the frequencies used by the two stations are different:

gNB:
```
Setting frequency: DL=1842.5 Mhz, DL_SSB=1843.25 Mhz (SSB-ARFCN=368650), UL=1747.5 MHz for cc_idx=0 nof_prb=52
```

UE:
```
2022-09-08T16:48:08.325429 [PHY-SA ] [I] [    0] Tuning Rx channel 0 to 1842.50 MHz
2022-09-08T16:48:08.325570 [RF     ] [I] Mapping RF channel 0 (device=0, channel=0) to logical carrier 0 on f_rx=1842.5 MHz
2022-09-08T16:48:08.327413 [PHY-SA ] [I] [    0] Cell search: Setting SSB configuration srate=11.52 MHz; c-freq=1842.500 MHz; ss-freq=1842.050 MHz; scs=15kHz; pattern=A; duplex=fdd;
```
Specifically the ssb channel of the UE seems different. You can try by playing around with the parameters: `dl_nr_arfcn` and `ssb_nr_arfcn` in the [rat.nr] section of ue.conf and the parameters: `dl_arfcn` and `ul_arfcn` in rr.conf

andrepuschmann, 2022-11-07:
Hi @nathanstep55, have you made progress with the issue meanwhile? In general I would recommend to use the 4G UE/eNB and calibrate your RF params before jumping to 5G. It'll help you to rule RF issues out.