srsran/srsRAN_4G #987: srsRAN compiling issue on aarch64 22.04

alphafox02, 2022-09-11:
## Issue Description ##
Building srsRAN from source on aarch64 22.04 fails 

## Expected Behavior ##
Compile finishes without error

## Actual Behaviour ##
`[  9%] Building C object lib/src/phy/phch/CMakeFiles/srsran_phch.dir/pdsch_nr.c.o                                                                               
In file included from /usr/src/srsRAN/lib/src/phy/fec/turbo/turbodecoder.c:102:
/usr/src/srsRAN/lib/include/srsran/phy/fec/turbo/turbodecoder_win.h: In function ‘tdec_winarm16_alpha.isra’:
/usr/src/srsRAN/lib/include/srsran/phy/fec/turbo/turbodecoder_win.h:313:44: error: ‘old[7]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  313 | #define int8x16_to_8x8x2(v) ((int8x8x2_t){{vget_low_s8(v), vget_high_s8(v)}}) // TODO
      |                                            ^~~~~~~~~~~
/usr/src/srsRAN/lib/include/srsran/phy/fec/turbo/turbodecoder_win.h:686:31: note: ‘old[7]’ was declared here
  686 |   simd_type_t m_b[8], new[8], old[8], max1[8], max0[8];
      |                               ^~~
/usr/src/srsRAN/lib/include/srsran/phy/fec/turbo/turbodecoder_win.h:313:44: error: ‘old[6]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  313 | #define int8x16_to_8x8x2(v) ((int8x8x2_t){{vget_low_s8(v), vget_high_s8(v)}}) // TODO
      |                                            ^~~~~~~~~~~
/usr/src/srsRAN/lib/include/srsran/phy/fec/turbo/turbodecoder_win.h:686:31: note: ‘old[6]’ was declared here
  686 |   simd_type_t m_b[8], new[8], old[8], max1[8], max0[8];
      |                               ^~~
/usr/src/srsRAN/lib/include/srsran/phy/fec/turbo/turbodecoder_win.h:313:44: error: ‘old[5]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  313 | #define int8x16_to_8x8x2(v) ((int8x8x2_t){{vget_low_s8(v), vget_high_s8(v)}}) // TODO
      |                                            ^~~~~~~~~~~
/usr/src/srsRAN/lib/include/srsran/phy/fec/turbo/turbodecoder_win.h:686:31: note: ‘old[5]’ was declared here
  686 |   simd_type_t m_b[8], new[8], old[8], max1[8], max0[8];
      |                               ^~~
/usr/src/srsRAN/lib/include/srsran/phy/fec/turbo/turbodecoder_win.h:313:44: error: ‘old[4]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  313 | #define int8x16_to_8x8x2(v) ((int8x8x2_t){{vget_low_s8(v), vget_high_s8(v)}}) // TODO
      |                                            ^~~~~~~~~~~
/usr/src/srsRAN/lib/include/srsran/phy/fec/turbo/turbodecoder_win.h:686:31: note: ‘old[4]’ was declared here
  686 |   simd_type_t m_b[8], new[8], old[8], max1[8], max0[8];
      |                               ^~~
/usr/src/srsRAN/lib/include/srsran/phy/fec/turbo/turbodecoder_win.h:313:44: error: ‘old[3]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  313 | #define int8x16_to_8x8x2(v) ((int8x8x2_t){{vget_low_s8(v), vget_high_s8(v)}}) // TODO
      |                                            ^~~~~~~~~~~
/usr/src/srsRAN/lib/include/srsran/phy/fec/turbo/turbodecoder_win.h:686:31: note: ‘old[3]’ was declared here
  686 |   simd_type_t m_b[8], new[8], old[8], max1[8], max0[8];
      |                               ^~~
/usr/src/srsRAN/lib/include/srsran/phy/fec/turbo/turbodecoder_win.h:313:44: error: ‘old[2]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  313 | #define int8x16_to_8x8x2(v) ((int8x8x2_t){{vget_low_s8(v), vget_high_s8(v)}}) // TODO
      |                                            ^~~~~~~~~~~
/usr/src/srsRAN/lib/include/srsran/phy/fec/turbo/turbodecoder_win.h:686:31: note: ‘old[2]’ was declared here
  686 |   simd_type_t m_b[8], new[8], old[8], max1[8], max0[8];
      |                               ^~~
/usr/src/srsRAN/lib/include/srsran/phy/fec/turbo/turbodecoder_win.h:313:44: error: ‘old[1]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  313 | #define int8x16_to_8x8x2(v) ((int8x8x2_t){{vget_low_s8(v), vget_high_s8(v)}}) // TODO
      |                                            ^~~~~~~~~~~
/usr/src/srsRAN/lib/include/srsran/phy/fec/turbo/turbodecoder_win.h:686:31: note: ‘old[1]’ was declared here
  686 |   simd_type_t m_b[8], new[8], old[8], max1[8], max0[8];
      |                               ^~~
/usr/src/srsRAN/lib/include/srsran/phy/fec/turbo/turbodecoder_win.h:313:44: error: ‘old[0]’ may be used uninitialized in this function [-Werror=maybe-uninitialized]
  313 | #define int8x16_to_8x8x2(v) ((int8x8x2_t){{vget_low_s8(v), vget_high_s8(v)}}) // TODO
      |                                            ^~~~~~~~~~~
/usr/src/srsRAN/lib/include/srsran/phy/fec/turbo/turbodecoder_win.h:686:31: note: ‘old[0]’ was declared here
  686 |   simd_type_t m_b[8], new[8], old[8], max1[8], max0[8];
      |  `

make: *** [Makefile:166: all] Error 2

## Steps to reproduce the problem ##
Attempt to compile with a fresh install of aarch64 22.04 for the Pi

andrepuschmann, 2022-09-21:
Hey, could you give [this](https://github.com/srsran/srsRAN/tree/agpl_next) branch I shot and see if it works for you?

Thanks

alphafox02, 2022-09-21:
That fixed it, thank you!

closed 2022-09-21 by alphafox02

zikkuratti, 2023-02-09:
@andrepuschmann @alphafox02 got the same issue what was at that link?