srsran/srsRAN_4G #989: Huawei P40 pro can search 5G SA cell but cannot finish RRC procedure

helloTkk, 2022-09-15:
Issue Description
Hello,

I followed this [note](https://docs.srsran.com/en/latest/app_notes/source/5g_sa_COTS/source/index.html) The srsenb and Open5GS works properly, the UE can see the 5G SA cell and send RACH messages, but cannot finish the RRC procedure as shown in the pcap. 

Setup Details
Operation System: Ubuntu 20.04
srsRAN 22.04
RF: USRP B210(with 2 VERT900 antennas)
UE: HUAWEI P40 pro
Card type: Sysmocom USIM(sysmoUSIM-SJS1)
Sim card configuration: 
Generated card parameters :
 > Name     : Magic
 > SMSP     : e1ffffffffffffffffffffffff0581005155f5ffffffffffff000000
 > ICCID    : 8988211000000452618
 > MCC/MNC  : 001/01
 > IMSI     : 901701234567895
 > Ki       : 00000000000000000000000064617665
 > OPC      : 64097b52589f63f12eec5172b49929d9
 > ACC      : None
 > ADM1(hex): 3439393633353439
 > OPMODE   : None

Expected Behavior
UE  connect to 5G SA network.

Actual Behaviour
UE can’t connect to 5G SA network, even cannot finish the RRC procedure. After RRC setup request and RRC setup message, without RRC setup complete, there are some RRC Resume request 1 messages. I have configured Wireshark as the enb.conf says, but Wireshark shows a Malformed packet and dissector bug as shown in the pcap.
[image]

Additional Information
I have several questions in this case: 
1. What is the wireshark version are used in this note? Does the wireshark needs any plug-in to decode 5G NR frames correctly?
2. Why the RRC setup complete message are missing, and RRC Resume request comes up?
3. Any help is appreciated!

iman432, 2022-09-16:
Hi,
Try to decrease nr_pdsch_mcs and nr_pusch_mcs, eg. 5 or 6 at enb.conf and also make sure using of USB 3 and CPU resources are enough.

[image]

helloTkk, 2022-09-16:
Hi @iman432, I tried your solution, it works for my Huawei P40 pro. Thanks a lot!

andrepuschmann, 2022-09-21:
@iman432 Thanks for the help.

@helloTkk Glad you got it working. I spotted a "malformed packet" coming from the phone in your PCAP. Would be interesting to see what the phone sent there.

Anyway, closing the issue as solved.

closed 2022-09-21 by andrepuschmann

bedrankara, 2022-09-30:
Hey @helloTkk , 

Does your Huawei have a chinese firmware version installed? I have one with a EU firmware and it seems like
5G SA is locked.