srsran/srsRAN_4G #990: OnePlus Nord 5G can't see 5G SA cell even using commercial plmn

helloTkk, 2022-09-15:
Issue Description
Hello,

I fully followed this [note](https://docs.srsran.com/en/latest/app_notes/source/5g_sa_COTS/source/index.html) The srsenb and Open5GS works properly, but my UE Oneplus Nord 5G can’t see the 5G SA gNB signal. I tried to set mmc and mnc as the same as the note, and I also tried set mmc and mnc to 001 01. None of the them works. I also tried to set a commercial mmc and mnc, and I failed again.

Setup Details
Operation System: Ubuntu 20.04
srsRAN 22.04
RF: USRP B210(with 2 VERT900 antennas)
UE: Oneplus Nord 5G
Card type: Sysmocom USIM(sysmoUSIM-SJA2)
Sim card configuration: 
Generated card parameters :
 > Name     : Magic
 > SMSP     : e1ffffffffffffffffffffffff0581005155f5ffffffffffff000000
 > ICCID    : 8988211000000452618
 > MCC/MNC  : 901/70
 > IMSI     : 901701234567895
 > Ki       : 00000000000000000000000064617665
 > OPC      : 64097b52589f63f12eec5172b49929d9
 > ACC      : None
 > ADM1(hex): 3439393633353439
 > OPMODE   : None

Expected Behavior
OnePlus Nord 5G detect the 5G SA network.

Actual Behaviour
OnePlus Nord 5G can’t search 5G SA network.

Steps to reproduce the problem
After running the gNB the NG connection is successful and the gNB is connected to the AMF. But my Oneplus Nord 5G can’t see it.

Additional Information
I have several questions in this case: 
1. Which antenna model are used in this note?
2. What is the model of the USIM are used in this note?
3. What is the the configuration for the USIM in this note? Can anyone share a script and configuration of USIM that works? I’m not sure whether it is the USIM problem leads to UE can’t see the 5G cell.
4. Any help is appreciated!

robertfalkenberg, 2022-09-19:
Hi,

first of all, since you use the sysmoUSIM-SJA2, have you already modified the 5G-related fields of the SIM card, as described [here](https://docs.srsran.com/en/latest/app_notes/source/5g_sa_COTS/source/index.html#g-sim)?

Regarding the other questions:

> Which antenna model are used in this note?

If the UE is close to the RF frontend, any simple dipole antenna that is suitable for the operating frequency should be OK.
In case the same setup works with 4G, then it's probably not the antennas.

> What is the model of the USIM are used in this note?

It is the sysmoSIM-SJS1 which is no longer available. However, numerous users confirmed it to work with sysmoUSIM-SJA2 after following the guide I linked above.

helloTkk, 2022-09-22:
Hi @robertfalkenberg,

Thanks for your reply! I am using sysmoUSIM-SJA2 and   tried this [note](https://gist.github.com/mrlnc/01d6300f1904f154d969ff205136b753). But when I excecute the command: pySIM-shell (MF/ADF.USIM/DF.5GS)> select EF.SUCI_Calc_Info there will be an error: EXCEPTION of type 'RuntimeError' occurred with message: '6283: Warnings - Selected file invalidated' 

Do you know in this case, what should I do? 

Btw, the link in your reply, am I supposed to try this command: ./pySim-shell.py -p0 --script ./scripts/deactivate-5g.script   to deactivate all the 5G related fields? or follow this [note](https://gist.github.com/mrlnc/01d6300f1904f154d969ff205136b753)?

Thanks again!

iacup, 2022-09-23:
The documentation is a little bit weird on that matter - the suggested script `./scripts/deactivate-5g.script` deactivates all 5G related files (incl. last PLMN, keys, etc.) and basically disables all 5G features on the sysmoUSIM-SJA2 card. Not sure why they suggest it. The [note](https://gist.github.com/mrlnc/01d6300f1904f154d969ff205136b753) serves only for enabling SUPI concealment (which AFAIK is not supported in current version of SRSRAN). As I said, it's weird.

The good thing is the phones mostly connect to 5G SA cell even with USIM cards which are old or have 5G features disabled. I observed that first five digits of IMSI matters. In my setup, phones cannot see any 5G SA cell if IMSI starts with 99970... or 90170.... If I change IMSI to start with 00101... phones suddenly see all of my 5G SA cells.

helloTkk, 2022-09-23:
Hi Jakub,

I tried a lot of combinations of IMSI including 00101, but it doesn’t work for my OnePlus Nord 5G… May I know what phones you are using?

helloTkk, 2022-09-26:
> Hi,
>
> first of all, since you use the sysmoUSIM-SJA2, have you already modified the 5G-related fields of the SIM card, as described [...]

Hi @robertfalkenberg,
Thanks for your reply. I have reprogramed my SIM card as the link you shared. But my Oneplus Nord 5G still cannot see the srsRAN 5G SA cell... If you are using Oneplus Nord 5G, could you please tell me what firware version you are using? Or you can connect to 5G with other COTS UE? By the way, I'm using usrp B210 without an GPSDO, is it important in this case?

Looking forward to your reply! Thanks again!

helloTkk, 2022-09-26:
> The documentation is a little bit weird on that matter - the suggested script `./scripts/deactivate-5g.script` deactivates all 5G related files (incl. last PLMN, keys, etc.) and basically disables [...]

Hi @iacup,
May I know what phone you are currently using to connect with srsRAN 5G SA cell? Thanks.

CassianoDev, 2022-09-26:
I have a POCO x4 pro and an iPhone13, but not can see the SA network created with srsRAN!
LTE network working normally with the same configuration.
I'm using a "Chinese OPENCELL" LTE USIM and it doesn't have the SUCI configuration, Is that why?
I'am use LimeSDR!

iacup, 2022-09-29:
> Hi @iacup, May I know what phone you are currently using to connect with srsRAN 5G SA cell? Thanks.

Hi, it's OnePlus N10 5G and ZTE Axon 30 5G. I had a lot of troubles finding 5G SA cell anyway. The IMSI turned out to be one of the key element in search for 5G but not the only one. I'm using B210 with _external clock_, the internal clock is not precise enough. I have also observed that lower TX gains increases chance of finding the cell (around 50-60). Also position of antenna and position of the phone on the table seems to matter. But here I'm getting into the realm of radio wave magic - which I do not understand at all. Maybe someone will explain. But once the phone finds and connects the 5G cell, it behaves a bit better.

helloTkk, 2022-10-03:
> 
@iacup Hi Iacup, thanks for your reply. I'm also using B210 but I haven't got my GPSDO yet. I have a question, after your phone connected to 5G SA cell, does your phone connected to the cell stable? In my experiment settings, I found that I have to set nr_pdsch_mcs = 3 and nr_pusch_mcs = 3 then my test phone Huawei p40 pro and Google Pixel 6 can connect to the cell. But they are not stable enough, they will disconnect and reconnect... I'm not sure whether this problem will be solved or not after I used GPSDO... May I know what's the value of nr_pdsch_mcs and nr_pdsch_mcs in your setteings? By the way, which antenna model did you use?(I'm using 2 VERT 900 with a usrp B210)

Looking forward to your reply! Thanks.

gabri94, 2022-10-06:
Hi, I'm using an Oneplus nord 5G AC 2003 and I can only see 5G networks with MCC 001 and MNC 02. I'm using opencells sim card programmed with their programmer.

During my tests i also tried to follow this procedure that might help you: https://mt-tech.fi/en/modify-cap_prune-to-change-5g-en_dc-sa-ca-combinations-on-qualcomm-5g-devices/

In my case all the relevant parameters were already set.

helloTkk, 2022-10-06:
Hi,

Thanks for your reply! I’ll further check the link you shared with me.

Btw, may I know your hardware settings?

Thanks again!

kbaccar, 2022-10-26:
Although on a different setup, I have tried the Oneplus Nord 2 on a setup composed of an Amarisoft Callbox working on 5G SA using band 3 and plmn 00101 and it worked without a problem. You might need to check the NR band that you are working on in your srsLTE config file, this list will give you a better idea about band availability for each phone https://cacombos.com/.
For your question about GPSDO and connection stability, i also had this problem connecting with srsUE, the problem was solved when I used a GPSDO on my B210.

helloTkk, 2022-10-31:
Hello,

I checked my Oneplus Nord 5G, it’s also AC2003 model.

May I know which model of Usrp you are using and could you please share your enb.conf file?

Thanks again!

iacup, 2022-11-02:
> @iacup Hi Iacup, thanks for your reply. I'm also using B210 but I haven't got my GPSDO yet. I have a question, after your phone connected to 5G SA cell, does your phone connected to the cell stable? [...]

Hi, yes, I had problems with phones on higher `nr_pd*ch_mcs`, today I'm using values around 10. I have observed that decreasing `tx_gain` significantly increases quality of signal on RX (maybe there is some self-interference). On my setup it holds: lower `tx_gain`, lower error rate. Furthermore I'm executing `srsenb` with `chrt -r 99` to increase priority and run the gNodeB in realtime. This also had significant impact on error rate and overall experience. I believe GPSDO (or external clock) is must in 5G. Seems like magic. I'm using small antenna just couple of inches tall, nothing special, not really good ones.

helloTkk, 2022-11-08:
> Hi, yes, I had problems with phones on higher `nr_pd*ch_mcs`, today I'm using values around 10. I have observed that decreasing `tx_gain` significantly increases quality of signal on RX (maybe there [...]

Thanks @iacup ! I am using usrp B210 and still cannot use nr_pdsch_mcs = 10 and nr_pusch_mcs = 10 to make the phone connect to it, so I guess there might be some other values have not been set correctly in my enb.conf file. Could you please share your enb.conf with me? My email is: tukai2259@gmail.com . Thanks again!

helloTkk, 2022-11-08:
> Hi, I'm using an Oneplus nord 5G AC 2003 and I can only see 5G networks with MCC 001 and MNC 02. I'm using opencells sim card programmed with their programmer.
> 
> During my tests i also tried to follow this procedure that might help you: https://mt-tech.fi/en/modify-oneplus-7-pro-5g-8-and-8-pro-nr-lte-a-band-combos/#Enable_5G_SA_Standalone_in_Android_11
> 
> In my case all the relevant parameters were already set and the thing that really changed was setting the the MCC and MNC in the sim (the first 5 number of the IMSI are now 00101)

Hi @gabri94 Thanks for your reply. I am also using Oneplus Nord 5G AC2003, and  I tried your solution, but it dosen't work for my UE. Could you please tell me which android version and firmware version you are using in your AC2003?