srsran/srsRAN_4G #991: ERROR_CODE_BAD_PACKET  was returned during streaming

fatafalihhilmi, 2022-09-16:
## Issue Description ##
when i run srsenb i got message a few seconds later 

## Setup Details ##
Ubuntu 16.04.07 LTS  i5-4300U CPU @ 1.90GHz dual core
UHD 3.9.2
srsran 21.10.0.

## Expected Behavior ##
running without error

## Actual Behaviour ##
---  Software Radio Systems LTE eNodeB  ---

Reading configuration file enb.conf...

Built in RelWithDebInfo mode using 21.10.0.

connect(): Connection refused
Failed to initiate S1 connection. Attempting reconnection in 10 seconds
Opening 1 channels in RF device=default with args=default
Available RF device list: UHD 
Trying to open RF device 'UHD'
Opening USRP channels=1, args: type=b200,master_clock_rate=23.04e6
-- Detected Device: B210
-- Operating over USB 3.
-- Initialize CODEC control...
-- Initialize Radio control...
-- Performing register loopback test... pass
-- Performing register loopback test... pass
-- Performing CODEC loopback test... pass
-- Performing CODEC loopback test... pass
-- Asking for clock rate 23.040000 MHz... 
-- Actually got clock rate 23.040000 MHz.
-- Performing timer loopback test... pass
-- Performing timer loopback test... pass
RF device 'UHD' successfully opened

==== eNodeB started ===
Type <t> to view trace
Setting frequency: DL=925.0 Mhz, UL=880.0 MHz for cc_idx=0 nof_prb=50
t
Enter t to stop trace.
/build/srsran-7ehIsQ/srsran-21.10/lib/src/phy/rf/rf_uhd_imp.cc:1349: Error ERROR_CODE_BAD_PACKET
 was returned during streaming. Aborting.
/build/srsran-7ehIsQ/srsran-21.10/lib/src/phy/rf/rf_uhd_imp.cc:1349: Error ERROR_CODE_BAD_PACKET
 was returned during streaming. Aborting.
/build/srsran-7ehIsQ/srsran-21.10/lib/src/phy/rf/rf_uhd_imp.cc:1349: Error ERROR_CODE_BAD_PACKET
 was returned during streaming. Aborting.
/build/srsran-7ehIsQ/srsran-21.10/lib/src/phy/rf/rf_uhd_imp.cc:1349: Error ERROR_CODE_BAD_PACKET
 was returned during streaming. Aborting.
/build/srsran-7ehIsQ/srsran-21.10/lib/src/phy/rf/rf_uhd_imp.cc:1349: Error ERROR_CODE_BAD_PACKET
 was returned during streaming. Aborting.
/build/srsran-7ehIsQ/srsran-21.10/lib/src/phy/rf/rf_uhd_imp.cc:1349: Error ERROR_CODE_BAD_PACKET
 was returned during streaming. Aborting.
/build/srsran-7ehIsQ/srsran-21.10/lib/src/phy/rf/rf_uhd_imp.cc:1349: Error ERROR_CODE_BAD_PACKET
 was returned during streaming. Aborting.
/build/srsran-7ehIsQ/srsran-21.10/lib/src/phy/rf/rf_uhd_imp.cc:1349: Error ERROR_CODE_BAD_PACKET
 was returned during streaming. Aborting.
/build/srsran-7ehIsQ/srsran-21.10/lib/src/phy/rf/rf_uhd_imp.cc:1349: Error ERROR_CODE_BAD_PACKET
 was returned during streaming. Aborting.
/build/srsran-7ehIsQ/srsran-21.10/lib/src/phy/rf/rf_uhd_imp.cc:1349: Error ERROR_CODE_BAD_PACKET
 was returned during streaming. Aborting.
/build/srsran-7ehIsQ/srsran-21.10/lib/src/phy/rf/rf_uhd_imp.cc:1349: Error ERROR_CODE_BAD_PACKET
 was returned during streaming. Aborting.
/build/srsran-7ehIsQ/srsran-21.10/lib/src/phy/rf/rf_uhd_imp.cc:1349: Error ERROR_CODE_BAD_PACKET
 was returned during streaming. Aborting.
/build/srsran-7ehIsQ/srsran-21.10/lib/src/phy/rf/rf_uhd_imp.cc:1349: Error ERROR_CODE_BAD_PACKET
 was returned during streaming. Aborting.
/build/srsran-7ehIsQ/srsran-21.10/lib/src/phy/rf/rf_uhd_imp.cc:1349: Error ERROR_CODE_BAD_PACKET
 was returned during streaming. Aborting.

robertfalkenberg, 2022-09-19:
Hi,
this issue is most probably fixed in more recent versions of srsRAN. Could you please upgrade to most recent version?
Moreover, we dropped support for Ubuntu 16.04; I suggest to upgrade to at least 20.04 LTS.

fatafalihhilmi, 2022-09-19:
@robertfalkenberg  hi, I've tried 20.04 LTS before but the error is = the [ERROR] [STREAMER] The receive packet handler caught a value exception. B210

do you have any suggestion what i should use?
Ubuntu 20.04 LTS
srsran ?
UHD ?

thanks

robertfalkenberg, 2022-09-20:
Yes, for example Ubuntu 20.04 LTS, latest srsRAN `master` branch [installed from source](https://docs.srsran.com/en/latest/general/source/1_installation.html#installation-from-source) and UHD 3.15.

closed 2022-09-20 by robertfalkenberg