srsran/srsRAN_4G #992: can't running epc

chatofking, 2022-09-19:
## Issue Description ##
[Describe the issue in detail]
can't running epc 
## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
pisoslite Debian 11 bulleye 
## Expected Behavior ##
[What you expect to happen]
can pass running epc step
## Actual Behaviour ##
[What happens instead e.g. error message]
 $ sudo ./srsepc/src/srsepc

```
Built in Release mode using commit ce8a3cae1 on branch master.

---  Software Radio Systems EPC  ---

Couldn't open , trying /root/.config/srsran/epc.conf
Couldn't open /root/.config/srsran/epc.conf either, trying /etc/srsran/epc.conf
Failed to read EPC configuration file /etc/srsran/epc.conf - exiting
```

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
https://docs.srsran.com/en/latest/app_notes/source/zeromq/source/index.html#running-the-epc
## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

robertfalkenberg, 2022-09-19:
Please check if the file `/root/.config/srsran/epc.conf` is present, since you run it as user `root` via `sudo`, or use the config install script as described [here](https://docs.srsran.com/en/latest/usermanuals/source/srsepc/source/2_epc_getstarted.html#getting-started)

chatofking, 2022-09-19:
how do can I download Debian script ??

chatofking, 2022-09-19:
yes is it  but i can't still run epc
chatofking@raspberrypi:~/.config/srsran $ srsran_install_configs.sh user
```
Installing srsRAN configuration files:
 - /home/chatofking/.config/srsran/ue.conf already exists. Skipping it.
 - /home/chatofking/.config/srsran/enb.conf already exists. Skipping it.
 - /home/chatofking/.config/srsran/sib.conf already exists. Skipping it.
 - /home/chatofking/.config/srsran/rr.conf already exists. Skipping it.
 - /home/chatofking/.config/srsran/rb.conf already exists. Skipping it.
 - /home/chatofking/.config/srsran/epc.conf already exists. Skipping it.
 - /home/chatofking/.config/srsran/mbms.conf already exists. Skipping it.
 - /home/chatofking/.config/srsran/user_db.csv already exists. Skipping it.
Done.
```

robertfalkenberg, 2022-09-19:
According to your output, the config files are located in the home directory of user `chatofking`. However, if you run `srsepc` as user `root` (via `sudo), the program `srsepc` searches the *home directory of the user that runs the application* for the config files.

If you run as `root`, it will search in `/root/.config/srsran`, not in `/home/chatofking/.config/srsran`.

You could copy the files to the other location and adjust your settings there.

closed 2022-09-19 by robertfalkenberg

chatofking, 2022-09-20:
when i dont use sugo it shows
chatofking@raspberrypi:~/srsRAN/build $ ./srsepc/src/srsepc

```
Built in Release mode using commit ce8a3cae1 on branch master.

---  Software Radio Systems EPC  ---

Couldn't open , trying /home/chatofking/.config/srsran/epc.conf
Reading configuration file /home/chatofking/.config/srsran/epc.conf...
Couldn't open user_db.csv, trying /home/chatofking/.config/srsran/user_db.csv
HSS Initialized.
MME S11 Initialized
MME GTP-C Initialized
MME Initialized. MCC: 0xf001, MNC: 0xff01
Could not initialize the SGi interface.
Could not initialize the SPGW's GTP-U.
Error initializing SP-GW
```

robertfalkenberg, 2022-09-20:
`srsEPC` need permissions to create a `tun` network device. This either requires `root` permissions, or the application binary needs special permissions to do it when run as a normal unprivileged user.
You can add the required permission flag as follows:
```
# enter build directory
cd ~/srsRAN/build/srsepc/src

# set cap_net_admin+eip flag
sudo setcap cap_net_admin+eip ./srsepc

# run the epc
./srsepc
```

jukomol, 2024-02-19:
> According to your output, the config files are located in the home directory of user `chatofking`. However, if you run `srsepc` as user `root` (via `sudo), the program `srsepc` searches the _home [...]

I am not sure which settings you are talking about. I am also facing the same problem. How to adjust the settings? Or how can I make  epc.conf accessible to srsepc?