srsran/srsRAN_4G #993: How to submit security issues?

takubokudori, 2022-09-20:
I've found multiple security issues on srsRAN, but there is no guideline about how to submit security issues in this repository.
Could you please tell me how to submit these?

andrepuschmann, 2022-09-20:
Hey @takubokudori,
thanks for getting in touch. We don't have a dedicated security process in place but that's something we will add in the future. Meanwhile, please feel free to send your findings to `codebot@srs.io` and we'll review them.

takubokudori, 2022-09-20:
@andrepuschmann

Thank you for the reply.
I've sent an e-mail to `security@srs.io`. Could you please check the e-mail box?

andrepuschmann, 2022-09-20:
We've received your mail. Thanks for the input

andrepuschmann, 2022-11-07:
For reference, we've also added a subsection under https://docs.srsran.com/en/latest/reporting_issues.html that explains the submission process for potential security issues. Thanks again for your contribution and pointing that out.

closed 2022-11-07 by andrepuschmann