srsran/srsRAN_4G #996: [ERROR] [UHD RF] EnvironmentError: IOError: usb rx6 transfer status: LIBUSB_TRANSFER_NO_DEVICE [ERROR] [UHD] An unexpected exception was caught in a task loop.The task loop will now exit, things may not work.EnvironmentError: IOError: usb rx8 transfer status: LIBUSB_TRANSFER_NO_DEVICE [ERROR] [UHD RF] EnvironmentError: IOError: usb rx6 transfer status: LIBUSB_TRANSFER_NO_DEVICE

fatafalihhilmi, 2022-09-20:
## Issue Description ##
when i run srsenb i got some error like this:
[ERROR] [UHD RF] EnvironmentError: IOError: usb rx6 transfer status: LIBUSB_TRANSFER_NO_DEVICE
[ERROR] [UHD] An unexpected exception was caught in a task loop.The task loop will now exit, things may not work.EnvironmentError: IOError: usb rx8 transfer status:LIBUSB_TRANSFER_NO_DEVICE
[ERROR] [UHD RF] EnvironmentError: IOError: usb rx6 transfer status: LIBUSB_TRANSFER_NO_DEVICE
[ERROR] [UHD RF] EnvironmentError: IOError: usb rx6 transfer status: LIBUSB_TRANSFER_NO_DEVICE
terminate called after throwing an instance of 'uhd::usb_error'
  what():  RuntimeError: USBError -4: usb tx2 submit failed: LIBUSB_ERROR_NO_DEVICE

## Setup Details ##
Ubuntu 20.04 LTS   i5-4300U CPU @ 1.90GHz 4 core 
srsran 22.04.1
UHD 3.15. LTS
USRP B210
## Expected Behavior ##
running without error
## Actual Behaviour ##
==== eNodeB started ===
Type <t> to view trace
Setting frequency: DL=2680.0 Mhz, UL=2560.0 MHz for cc_idx=0 nof_prb=50
t
Enter t to stop trace.
[ERROR] [UHD RF] EnvironmentError: IOError: usb rx6 transfer status: LIBUSB_TRANSFER_NO_DEVICE
[ERROR] [UHD] An unexpected exception was caught in a task loop.The task loop will now exit, things may not work.EnvironmentError: IOError: usb rx8 transfer status: LIBUSB_TRANSFER_NO_DEVICE
[ERROR] [UHD RF] EnvironmentError: IOError: usb rx6 transfer status: LIBUSB_TRANSFER_NO_DEVICE
[ERROR] [UHD RF] EnvironmentError: IOError: usb rx6 transfer status: LIBUSB_TRANSFER_NO_DEVICE
terminate called after throwing an instance of 'uhd::usb_error'
  what():  RuntimeError: USBError -4: usb tx2 submit failed: LIBUSB_ERROR_NO_DEVICE
--- command='srsenb enb.conf' version=22.04.1 signal=6 date='20/09/2022 09:21:36' ---
	srsenb(+0x35ec37) [0x55d31c668c37]
	/lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f163501e210]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcb) [0x7f163501e18b]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x12b) [0x7f1634ffd859]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0x9e911) [0x7f16353d2911]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xaa38c) [0x7f16353de38c]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xa9369) [0x7f16353dd369]
	/lib/x86_64-linux-gnu/libstdc++.so.6(__gxx_personality_v0+0x2a1) [0x7f16353ddd21]
	/lib/x86_64-linux-gnu/libgcc_s.so.1(+0x10bef) [0x7f16351dabef]
	/lib/x86_64-linux-gnu/libgcc_s.so.1(_Unwind_Resume+0x12a) [0x7f16351db5aa]
	/usr/local/lib/libuhd.so.3.15.0(+0x96cf13) [0x7f1634b2af13]
	/usr/local/lib/libuhd.so.3.15.0(+0x96add9) [0x7f1634b28dd9]
	/usr/local/lib/libuhd.so.3.15.0(+0x3a4ffd) [0x7f1634562ffd]
	/usr/local/lib/libuhd.so.3.15.0(+0x702569) [0x7f16348c0569]
	/usr/local/lib/libsrsran_rf_uhd.so(+0x263d3) [0x7f1634e8a3d3]
	/usr/local/lib/libsrsran_rf_uhd.so(rf_uhd_send_timed_multi+0x51a) [0x7f1634e7867a]
	/usr/local/lib/libsrsran_rf_uhd.so(rf_uhd_send_timed+0x5d) [0x7f1634e790ed]
	/usr/local/lib/libsrsran_rf.so.0(srsran_rf_send_timed2+0x2d) [0x7f163572fdbd]
	srsenb(+0x54c632) [0x55d31c856632]
	srsenb(+0x54cc13) [0x55d31c856c13]
	srsenb(+0xd427b) [0x55d31c3de27b]
	srsenb(+0xe1456) [0x55d31c3eb456]
	srsenb(+0x3766e2) [0x55d31c6806e2]
	srsenb(+0xa9e25) [0x55d31c3b3e25]
	/lib/x86_64-linux-gnu/libpthread.so.0(+0x9609) [0x7f1635873609]
	/lib/x86_64-linux-gnu/libc.so.6(clone+0x43) [0x7f16350fa293]
srsRAN crashed. Please send this backtrace to the developers ...
---  exiting  ---
terminate called recursively
--- command='srsenb enb.conf' version=22.04.1 signal=6 date='20/09/2022 09:21:36' ---
	srsenb(+0x35ec37) [0x55d31c668c37]
	/lib/x86_64-linux-gnu/libc.so.6(+0x46210) [0x7f163501e210]
	/lib/x86_64-linux-gnu/libc.so.6(gsignal+0xcb) [0x7f163501e18b]
	/lib/x86_64-linux-gnu/libc.so.6(abort+0x12b) [0x7f1634ffd859]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xac53a) [0x7f16353e053a]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xaa38c) [0x7f16353de38c]
	/lib/x86_64-linux-gnu/libstdc++.so.6(+0xaa3f7) [0x7f16353de3f7]
	/usr/local/lib/libsrsran_rf_uhd.so(+0x194a8) [0x7f1634e7d4a8]
	/usr/local/lib/libsrsran_rf_uhd.so(+0x443b2) [0x7f1634ea83b2]
	/lib/x86_64-linux-gnu/libc.so.6(__cxa_finalize+0xce) [0x7f163502215e]
	/usr/local/lib/libsrsran_rf_uhd.so(+0x11647) [0x7f1634e75647]
srsRAN crashed. Please send this backtrace to the developers ...
---  exiting  ---

## Steps to reproduce the problem ##
change generic kernel to low latency kernel
set CPU GOVERNOR=performance

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

fatafalihhilmi, 2022-09-20:
@robertfalkenberg

robertfalkenberg, 2022-09-21:
This
```
terminate called after throwing an instance of 'uhd::usb_error'
what(): RuntimeError: USBError -4: usb tx2 submit failed: LIBUSB_ERROR_NO_DEVICE
```
is an error message coming from the UHD driver. Please verify proper installation and function of the driver and the device.
https://kb.ettus.com/Verifying_the_Operation_of_the_USRP_Using_UHD_and_GNU_Radio
Please also check the USB connection cable (e.g. loose connection).

closed 2022-11-07 by andrepuschmann