srsran/srsRAN_4G #998: LimeSDR + LimeRFE + srsRAN

egortaran, 2022-09-22:
## Issue Description
Hello everyone. I would like to run srsRAN on LimeSDR v1.4s using LimeRFE amplifier. How to do it?
I use this [documentation](https://github.com/myriadrf/LimeRFE/tree/master/hardware/1v31/docs). When I use LimeSuite GUI, following the instructions, the ports open. 

<details>
  <summary>LimeSDR</summary>

  ```
INFO: Disconnected control port
INFO: Reference clock 30.72 MHz
INFO: Connected Control port: LimeSDR-USB FW:4 HW:4 Protocol:1 GW:2.23 Ref Clk: 30.72 MHz
  ```
</details>

<details>
  <summary>LimeRFE</summary>

  ```
Port opened
Firmware version: 1
Hardware version: 0x10
  ```
</details>

But since the device is already busy, I can not be able to connect srsRAN.

<details>
  <summary>srsRAN</summary>

  ```
[ERROR] Cannot claim interface - Resource busy
[ERROR] Failed to open device
[ERROR] connection is not open
[ERROR] TransferPacket: Write failed (ret=0)
[ERROR] connection is not open
[ERROR] TransferPacket: Write failed (ret=0)
[ERROR] Failed to open. Device is busy.
Failed to create Soapy object
Error initializing radio.
  ```
</details>

## Technology stack
Ubuntu 22.04 (Intel® Core™ i7-7700 CPU @ 3.60GHz × 8)
srsRAN 22.04 (open5gs)
LimeSDR v1.4s (LimeSuite v20.10.0, SoapySDR 0.7.2)
LimeRFE 1v31

andrepuschmann, 2022-09-22:
I am not sure how their interfacing works, but that's a Lime issue rather than srsRAN I'd say. Have you checked their pages yet?

jtang613, 2022-09-24:
It looks like LimeSuite is connected to your LimeSDR, causing contention when SRSRAN is trying to claim the SDR interface. Or possibly it is mistakenly detecting LimeRFE as a LimeSDR - which you can avoid by explicitly adding the "id=n" option to your soapy device args string. If using the LimeRFE, you'll also want to ensure that you're using the "Direct (USB)" connection as opposed to the SDR connection and disconnect the LimeSDR from LimeSuite so that the SDR interface remains available to SRSRAN. Keep in mind that SRSRAN is somewhat hardware-agnostic, so it won't have builtin support for hardware like the LimeRFE - you'll need to configure the LimeRFE frequency and power levels manually to match what you have configured SRSRAN to operate on.

andrepuschmann, 2022-11-07:
Closing this ticket as not feedback has been provided by the original author.

closed 2022-11-07 by andrepuschmann