srsran/srsRAN_4G #1001: RACH procedure failed for Quactel module

orion-belt, 2022-09-26:
## Issue Description ##
RACH procedure failed

## Setup Details ##
* srsenb running inside pod and using USRP X310, 
* UHD version 4.0
* Worker details
```bash
CPU(s):                          128
On-line CPU(s) list:             0-127
Thread(s) per core:              2
Core(s) per socket:              32
Model name:                      Intel(R) Xeon(R) Gold 6338 CPU @ 2.00GHz
Stepping:                        6
CPU MHz:                         2000.000
```

## Expected Behavior ##
RACH procedure success

## Actual Behaviour ##
PLMN (001,01) detected at UE (Quactel module)
```bash
2022-09-25 22:40:50: [SEQ96,ID72]>>> at+cops=?
2022-09-25 22:41:00: [SEQ0,ID0]<<< +COPS: (1,"Test PLMN 1-1","Test1-1","00101",12),(1,"313 100","313 100","313100",7),(1,"AT&T","AT&T","310410",7),,(0-4),(0-2)
```
Gnb logs 
```bash
root@srsran-gnb-7bc85d9cbb-f65s8:/srsran/bin# ./srsenb 
Active RF plugins: libsrsran_rf_uhd.so libsrsran_rf_zmq.so
Inactive RF plugins: 
---  Software Radio Systems LTE eNodeB  ---

Couldn't open , trying /root/.config/srsran/enb.conf
Reading configuration file /root/.config/srsran/enb.conf...
Couldn't open sib.conf, trying /root/.config/srsran/sib.conf
Couldn't open rr.conf, trying /root/.config/srsran/rr.conf
Couldn't open rb.conf, trying /root/.config/srsran/rb.conf

Built in Release mode using commit ce8a3cae1 on branch HEAD.

Opening 1 channels in RF device=default with args=fail_on_disconnect=true,tx_port=tcp://*:2000,rx_port=tcp://localhost:2001,id=enb,base_srate=11.52e6
Supported RF device list: UHD zmq file
Trying to open RF device 'UHD'
NG connection successful
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_4.0.0.HEAD-0-g90ce6062
[INFO] [LOGGING] Fastpath logging disabled at runtime.
Opening USRP channels=1, args: fail_on_disconnect=true,tx_port=tcp://*:2000,rx_port=tcp://localhost:2001,id=enb,base_srate=11.52e6,type=x300,master_clock_rate=184.32e6
[INFO] [UHD RF] RF UHD Generic instance constructed
[INFO] [X300] X300 initialization sequence...
[INFO] [X300] Maximum frame size: 8000 bytes.
[INFO] [X300] Radio 1x clock: 184.32 MHz
RF device 'UHD' successfully opened

==== eNodeB started ===
Type <t> to view trace
Setting frequency: DL=1842.5 Mhz, DL_SSB=1842.05 Mhz (SSB-ARFCN=368410), UL=1747.5 MHz for cc_idx=0 nof_prb=52

RACH:  slot=971, cc=0, preamble=3, offset=9, temp_crnti=0x4601
Disconnecting rnti=0x4601.
RACH:  slot=1131, cc=0, preamble=0, offset=31, temp_crnti=0x4602
Disconnecting rnti=0x4602.
RACH:  slot=1291, cc=0, preamble=7, offset=45, temp_crnti=0x4603
Disconnecting rnti=0x4603.
RACH:  slot=1451, cc=0, preamble=6, offset=62, temp_crnti=0x4604
Disconnecting rnti=0x4604.

```
## Steps to reproduce the problem ##

## Additional Information ##

andrepuschmann, 2022-11-07:
Could be related to https://github.com/srsran/srsRAN/issues/1010. Could you please send logs/PCAPs?

plaiddshirt, 2024-01-30:
Hi, may I ask which model of Quectel's module you are using, I am also making an attempt to access srs_RAN base station via Quectel module, however the EC20 I have does not support this connection. Thanks a lot!