srsran/srsRAN_4G #1002: Can't doing N2 Handover using srsue, srsenb, zmq, open5gs

fajarmhr, 2022-09-29:
Hi everyone, i'm new here.

i'm trying to do n2 handover by modifying steps from [S1 Handover](https://docs.srsran.com/en/latest/app_notes/source/handover/source/index.html#s1-handover) and use NR steps from [5G SA End-to-End](https://docs.srsran.com/en/latest/app_notes/source/5g_sa_E2E/source/index.html) but it still doesn't work. 

did I miss something?

## Setup Details ##
this is attach of my config file
[n2handov.zip](https://github.com/srsran/srsRAN/files/9671760/n2handov.zip)
[5Gcore.zip](https://github.com/srsran/srsRAN/files/9671791/5Gcore.zip)

## Actual Behaviour ##
[image]
when I slide up the neighbor cell gain (cell0) and decrease the source cell (cell1), RSRP weakens but does not switch cells

fajarmhr, 2022-09-29:
Here is the amf log

09/29 14:47:33.352: [amf] INFO: gNB-N2 accepted[127.0.1.1]:55194 in ng-path module (../src/amf/ngap-sctp.c:113)
09/29 14:47:33.352: [amf] INFO: gNB-N2 accepted[127.0.1.1] in master_sm module (../src/amf/amf-sm.c:659)
09/29 14:47:33.352: [amf] INFO: [Added] Number of gNBs is now 1 (../src/amf/context.c:880)
09/29 14:47:33.352: [amf] INFO: gNB-N2[127.0.1.1] max_num_of_ostreams : 30 (../src/amf/amf-sm.c:698)
09/29 14:47:41.681: [amf] INFO: gNB-N2 accepted[127.0.1.2]:45192 in ng-path module (../src/amf/ngap-sctp.c:113)
09/29 14:47:41.681: [amf] INFO: gNB-N2 accepted[127.0.1.2] in master_sm module (../src/amf/amf-sm.c:659)
09/29 14:47:41.681: [amf] INFO: [Added] Number of gNBs is now 2 (../src/amf/context.c:880)
09/29 14:47:41.681: [amf] INFO: gNB-N2[127.0.1.2] max_num_of_ostreams : 30 (../src/amf/amf-sm.c:698)
09/29 14:47:55.982: [amf] INFO: InitialUEMessage (../src/amf/ngap-handler.c:361)
09/29 14:47:55.982: [amf] INFO: [Added] Number of gNB-UEs is now 1 (../src/amf/context.c:2132)
09/29 14:47:55.982: [amf] INFO: Unknown UE by 5G-S_TMSI[AMF_ID:0x20000,M_TMSI:0x79714f0d] (../src/amf/ngap-handler.c:431)
09/29 14:47:55.982: [amf] INFO:     RAN_UE_NGAP_ID[1] AMF_UE_NGAP_ID[9] TAC[7] CellID[0x670001] (../src/amf/ngap-handler.c:497)
09/29 14:47:55.982: [amf] INFO: [suci-0-999-70-0000-0-0-0123456780] known UE by SUCI (../src/amf/context.c:1441)
09/29 14:47:55.982: [gmm] INFO: Registration request (../src/amf/gmm-sm.c:135)
09/29 14:47:55.982: [gmm] INFO: [suci-0-999-70-0000-0-0-0123456780]    SUCI (../src/amf/gmm-handler.c:149)
09/29 14:47:55.982: [amf] INFO: [imsi-999700123456780:1] Release SM context [204] (../src/amf/amf-sm.c:467)
09/29 14:47:55.982: [amf] INFO: [Removed] Number of AMF-Sessions is now 0 (../src/amf/context.c:2153)
09/29 14:47:56.693: [gmm] INFO: [imsi-999700123456780] Registration complete (../src/amf/gmm-sm.c:1124)
09/29 14:47:56.693: [amf] INFO: [imsi-999700123456780] Configuration update command (../src/amf/nas-path.c:430)
09/29 14:47:56.693: [gmm] INFO:     UTC [2022-09-29T07:47:56] Timezone[0]/DST[0] (../src/amf/gmm-build.c:531)
09/29 14:47:56.693: [gmm] INFO:     LOCAL [2022-09-29T14:47:56] Timezone[25200]/DST[0] (../src/amf/gmm-build.c:536)
09/29 14:47:56.693: [amf] INFO: [Added] Number of AMF-Sessions is now 1 (../src/amf/context.c:2146)
09/29 14:47:56.693: [gmm] INFO: UE SUPI[imsi-999700123456780] DNN[internet] S_NSSAI[SST:1 SD:0xffffff] (../src/amf/gmm-handler.c:1062)
09/29 14:47:56.693: [sbi] WARNING: Try to discover [nnssf-nsselection] (../lib/sbi/path.c:114)
09/29 14:47:56.693: [sbi] WARNING: [1ac709a4-3fa6-41ed-bd75-952c1f59bbc1] (NF-discover) NF has already been added (../lib/sbi/nnrf-handler.c:735)
09/29 14:47:56.693: [sbi] WARNING: NF EndPoint updated [127.0.0.14:80] (../lib/sbi/context.c:1564)
09/29 14:47:56.693: [sbi] WARNING: NF EndPoint updated [127.0.0.14:7777] (../lib/sbi/context.c:1473)
09/29 14:47:56.693: [sbi] INFO: [1ac709a4-3fa6-41ed-bd75-952c1f59bbc1] (NF-discover) NF Profile updated (../lib/sbi/nnrf-handler.c:758)
09/29 14:47:56.693: [amf] WARNING: Try to discover [nsmf-pdusession] (../src/amf/sbi-path.c:339)
09/29 14:47:56.694: [sbi] WARNING: [1abfa1b4-3fa6-41ed-8533-03884d0475f6] (NF-discover) NF has already been added (../lib/sbi/nnrf-handler.c:735)
09/29 14:47:56.694: [sbi] WARNING: NF EndPoint updated [127.0.0.4:80] (../lib/sbi/context.c:1564)
09/29 14:47:56.694: [sbi] WARNING: NF EndPoint updated [127.0.0.4:7777] (../lib/sbi/context.c:1473)
09/29 14:47:56.694: [sbi] INFO: [1abfa1b4-3fa6-41ed-8533-03884d0475f6] (NF-discover) NF Profile updated (../lib/sbi/nnrf-handler.c:758)
09/29 14:47:57.037: [gmm] ERROR: [imsi-999700123456780] No GUTI allocated (../src/amf/gmm-sm.c:346)

andrepuschmann, 2022-11-07:
Mobility isn't supported for 5G, only 4G.

closed 2022-11-07 by andrepuschmann

Moussa-Guemdani, 2024-01-18:
Hello,

Does it now support mobility for 5G, and if so, does it implicitly require the Xn interface?

Thank you.