srsran/srsRAN_4G #1003: srsUE can't connect to MagmaCore EPC - Attach failure

albukirky1, 2022-10-04:
## Issue Description ##
I'm trying to attach srsUE to srsENB to [Magma's ](https://github.com/magma/magma) AGW (EPC).  
Currently I've managed, what appears to, successfully connect srsenb to the EPC, but when trying to initiate the srsue in the environment, the attach process fails (5 retries are attempted before disregarding), and there is no communication between the UE and the EPC (i.e. pinging uplink does not work nor downlink because no IP is assigned to the UE).  

My biggest uncertainty is the configuration of the UE and the ENB (in Additional Information section). I'm feeling like I might be configuring something wrong, network-wise, because the exact configuration worked for this [repo](https://github.com/ShubhamTatvamasi/srsRAN-demo) with Magma.

## Setup Details ##
Operating System: ubuntu-20.04.3-live-server-amd64  
This same OS is used for both Magma's EPC (which is running on a VMware VM) and the srsenb+srsue (which run together on another VM).  
I'm using bridged network for both of the VM's and they can ping each other with no problem,
<details>
  <summary>EPC network setup</summary>

        dhcp0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
                inet6 fe80::c06c:32ff:fea8:a94f  prefixlen 64  scopeid 0x20<link>
                ether c2:6c:32:a8:a9:4f  txqueuelen 1000  (Ethernet)
                RX packets 23  bytes 1628 (1.6 KB)
                RX errors 0  dropped 0  overruns 0  frame 0
                TX packets 20  bytes 1496 (1.4 KB)
                TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        
        envoy_cntr: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
                inet 10.5.0.2  netmask 255.255.255.0  broadcast 10.5.0.255
                inet6 fe80::50f5:a2ff:fe22:4854  prefixlen 64  scopeid 0x20<link>
                ether 52:f5:a2:22:48:54  txqueuelen 1000  (Ethernet)
                RX packets 470  bytes 46859 (46.8 KB)
                RX errors 0  dropped 0  overruns 0  frame 0
                TX packets 468  bytes 26630 (26.6 KB)
                TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        
        eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
                inet 10.0.0.110  netmask 255.255.255.0  broadcast 10.0.0.255
                inet6 2a06:c701:9d7d:e900:20c:29ff:fe1e:ca94  prefixlen 64  scopeid 0x0<global>
                inet6 fe80::20c:29ff:fe1e:ca94  prefixlen 64  scopeid 0x20<link>
                ether 00:0c:29:1e:ca:94  txqueuelen 1000  (Ethernet)
                RX packets 158494  bytes 13432697 (13.4 MB)
                RX errors 2  dropped 2354  overruns 0  frame 0
                TX packets 56630  bytes 25321589 (25.3 MB)
                TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
                device interrupt 19  base 0x2000
        
        eth1: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
                inet 10.0.0.120  netmask 255.255.255.0  broadcast 10.0.0.255
                inet6 2a06:c701:9d7d:e900:20c:29ff:fe1e:ca9e  prefixlen 64  scopeid 0x0<global>
                inet6 fe80::20c:29ff:fe1e:ca9e  prefixlen 64  scopeid 0x20<link>
                ether 00:0c:29:1e:ca:9e  txqueuelen 1000  (Ethernet)
                RX packets 131823  bytes 10018662 (10.0 MB)
                RX errors 1  dropped 2354  overruns 0  frame 0
                TX packets 14038  bytes 2031272 (2.0 MB)
                TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
                device interrupt 16  base 0x2080
        
        gtp_br0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
                inet 192.168.128.1  netmask 255.255.255.0  broadcast 192.168.128.255
                inet6 fe80::9ceb:c5ff:fe64:4540  prefixlen 64  scopeid 0x20<link>
                ether 9e:eb:c5:64:45:40  txqueuelen 1000  (Ethernet)
                RX packets 15  bytes 1064 (1.0 KB)
                RX errors 0  dropped 3  overruns 0  frame 0
                TX packets 19  bytes 1426 (1.4 KB)
                TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
        
        gtpu_sys_2152: flags=4305<UP,POINTOPOINT,RUNNING,NOARP,MULTICAST>  mtu 65499
                inet6 fe80::3ed1:cfbf:352d:9320  prefixlen 64  scopeid 0x20<link>
                unspec 00-00-00-00-00-00-00-00-00-00-00-00-00-00-00-00  txqueuelen 1000  (UNSPEC)
                RX packets 0  bytes 0 (0.0 B)
                RX errors 0  dropped 0  overruns 0  frame 0
                TX packets 0  bytes 0 (0.0 B)
                TX errors 15  dropped 0 overruns 0  carrier 0  collisions 0
        
        ipfix0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
[... 438 lines cut ...]
    2022-10-04T02:44:01.085948 [S1AP   ] [E] IPv6 addresses not currently supported
    2022-10-04T02:44:38.148261 [S1AP   ] [E] IPv6 addresses not currently supported
    2022-10-04T02:45:14.856938 [S1AP   ] [E] IPv6 addresses not currently supported
    2022-10-04T02:45:52.507862 [S1AP   ] [E] IPv6 addresses not currently supported
    2022-10-04T02:47:55.366666 [COMN   ] [D] [    0] RxSockets: Closing rx socket handler thread
    2022-10-04T02:47:55.367388 [COMN   ] [D] [    0] RxSockets: closed.

</details>

* provided INFO because DEBUG was enormous. Will provide DEBUG if necessary.

<details>
  <summary>ue.log</summary>

    2022-10-04T02:43:22.629649 [UE     ] [I] Using binary srsue/src/srsue with arguments:
    2022-10-04T02:43:22.669845 [UE     ] [I] Built in Release mode using commit ce8a3cae1 on branch master.
    2022-10-04T02:43:45.678478 [NAS    ] [W] Timer T3410 expired after attach attempt 1/5: starting T3411
    2022-10-04T02:44:22.743967 [NAS    ] [W] Timer T3410 expired after attach attempt 2/5: starting T3411
    2022-10-04T02:44:59.571928 [NAS    ] [W] Timer T3410 expired after attach attempt 3/5: starting T3411
    2022-10-04T02:45:37.244090 [NAS    ] [W] Timer T3410 expired after attach attempt 4/5: starting T3411
    2022-10-04T02:46:14.025145 [NAS    ] [W] Timer T3410 expired. Maximum attempts reached. Starting T3402

</details>

* provided INFO because DEBUG was enormous. Will provide DEBUG if necessary.

andrepuschmann, 2022-11-07:
@albukirky1 Please excuse the long delay in answering. I appreciate your very detailed and excellent issue report. The UE failing to attach is simply because the eNB didn't complete the S1 registration with the core. So it will reject each UE. Please check in the EPC/eNB logs and configuration and make sure the S1 setup completes successfully.

im-nonedone, 2024-09-08:
@albukirky1 did you ever find out what the issue was?

lucaaamaral, 2025-03-14:
I'm having a similar issue. I had commented over the Magma issues https://github.com/magma/magma/issues/14074 .
This is annoying me, because srsRAN_4G was my default tool to debug a core for over two years and I have just recently switched to magma, and I can't get 4G to work with srsRAN.