srsran/srsRAN_4G #1004: Couldn't run enodeB

Mogit21, 2022-10-04:
## Issue Description ##
[Cannot run enodeB]
Cannot run enodeB
## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
full end-to-end LTE network on a single computer
Operation System: 9.16.1-Ubuntu

## Expected Behavior ##
[enodeB should run]

## Actual Behaviour ##
[What happens instead e.g. error message]
haziane@haziane-Inspiron-15-3567:~$ ./srsenb/src/srsenb --rf.device_name=zmq --rf.device_args="fail_on_disconnect=true,tx_port=tcp://*:2000,rx_port=tcp://localhost:2001,id=enb,base_srate=23.04e6"
bash: ./srsenb/src/srsenb: No such file or directory
haziane@haziane-Inspiron-15-3567:~$ sudo srsenb
---  Software Radio Systems LTE eNodeB  ---

Couldn't open , trying /root/.config/srsran/enb.conf
Reading configuration file /root/.config/srsran/enb.conf...
Couldn't open sib.conf, trying /root/.config/srsran/sib.conf
Couldn't open rr.conf, trying /root/.config/srsran/rr.conf
Couldn't open rb.conf, trying /root/.config/srsran/rb.conf
WARNING: cpu0 scaling governor is not set to performance mode. Realtime processing could be compromised. Consider setting it to performance mode before running the application.

Built in RelWithDebInfo mode using 22.04.1.

Optional 5G SRB configuration is not supported yet.
Please specify 5G SRB1 and SRB2 configuration.
Error parsing RB configuration
Error processing arguments.
## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]
Followed the steps in srsRAN with ZMQ Virtual Radios, link https://docs.srsran.com/en/latest/app_notes/source/zeromq/source/index.html#srsran-with-zmq-virtual-radios
## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

jtang613, 2022-10-08:
Looks like the type of error you'd see from running a recent release with outdated config files. Try replacing your old config files with the new versions in the source tree.

andrepuschmann, 2022-11-07:
And again, thanks @jtang613

closed 2022-11-07 by andrepuschmann