srsran/srsRAN_4G #1006: Does the srsRAN zmq support connection of two computers?

LongRiver-hash, 2022-10-08:
## Issue Description ##
I tried to use srsran zmq to connect two computers，the ue stopped at “Attaching UE...”.It worked on one computer.So I want to know does the srsRAN zmq support connection of two computers?(one srsepc,srsenb ，another srsue)，if support，how to write config files？thank you very much！ 

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

jtang613, 2022-10-08:
In the docs: https://docs.srsran.com/en/latest/app_notes/source/zeromq/source/index.html#known-issues
"We currently only support a single eNB and a single UE."

LongRiver-hash, 2022-10-09:
I have received the docs: https://docs.srsran.com/en/latest/app_notes/source/zeromq/source/index.html#known-issues.
But there is a question. "Support a single eNB and a single UE."means on one computer?How can I use srsran zmq to connect two computers?(one computer a single eNB,another computer a single UE.Becasue we know the USRP need two computers.)

jtang613, 2022-10-09:
Yes, in this case, it's simply a matter of entering the remote IP addresses in the zmq device_args line in both the enb.conf and ue.conf:
```device_args = fail_on_disconnect=false,tx_port=tcp://*:2000,rx_port=tcp://<remote_ip>:2001,id=enb,base_srate=11.52e6```

andrepuschmann, 2022-11-07:
Thanks @jtang613 for providing the right answer. Hope that helped you @LongRiver-hash. Closing the issue.

closed 2022-11-07 by andrepuschmann