srsran/srsRAN_4G #1008: how to craft a subframe using srsRAN library

Chico233, 2022-10-08:
## Issue Description ##
I want to craft a subframe, do you guys know how I could do that with srsRAN library?
Thanks!!!
## Setup Details ##
USRP B210, Ubuntu 18.04.6 LTS, srsRAN_release_21_10

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

zikkuratti, 2022-10-10:
craft like beer you mean?)

zikkuratti, 2022-10-10:
if not) you can search variable you want to reconfigure in IDE and add some values actual for you reglament and 3GPP doc's.

andrepuschmann, 2022-11-07:
Closing the ticket as this isn't an issue with the code-base itself. Please use the mailing list for this sort of discussions.

closed 2022-11-07 by andrepuschmann