srsran/srsRAN_4G #1009: bladeRF cannot be opened

jbarea, 2022-10-11:
I compiled and built the program as stated in the docs. Also modified the config files accordingly, but everytime i launch sudo srsue
i get an error:
Opening 1 channels in RF device=bladeRF with args=""
Supported RF device list: UHD file
/build/srsran-WN5iNl/srsran-22.04.1/lib/src/phy/rf/rf_imp.c:129: RF device 'bladeRF' not found. Please check the available srsRAN CMAKE options to verify if this device is being detected in your system
Error initializing radio.
 
I have the bladerf driver installed as well as libraries needed, as i have othe programs using it correctly. Is anyone experiencing this type of error?
Thanks in advance!!!

fllay, 2022-10-12:
You may need to rebuild srsran after BladeRF driver installation. You can see from the output that BladeRF  device is not included in your srsran.

`Supported RF device list: UHD file`

If you srsran is build with bladeRF driver support, you will see something like this
 
`Supported RF device list: UHD soapy bladeRF file`

jbarea, 2022-10-14:
BladeRf driver was installed when i built srsran, as i have been using other software that uses the bladerf. The libraries and the dev packages were installed also. Is there a special argument when building srsran in order to tell the bladerf support?.
Thanks for the reply!

fllay, 2022-10-15:
I installed everything from source. You may try to install everything from source including BladeRF driver.

andrepuschmann, 2022-11-07:
Has the issue been solved meanwhile? I guess it's a compilation/built issue as has been pointed out by others. Closing the ticket but please feel free to reopen if it still doesn't work for you.

closed 2022-11-07 by andrepuschmann

wesmwoo, 2023-01-10:
I'm also encountering this issue -- has anyone resolved this?