srsran/srsRAN_4G #1010: COTS UE can't connect

Salous-99, 2022-10-12:
## Issue Description ##
COTS UE cannot establish an RRC connection with the gNB. The console displays that multiple RACH are established and subsequently time out and are dropped. Upon analyzing the dumped pcap file, I see some malformed packets which, to my understanding, means that there is most probably an issue with the configuration at the physical layer. Eventually all RACHs are dropped and nothing happens.

## Setup Details ##
Core network: Open5GS running on Ubuntu 20.04
gNB: srsenb
CPU: 11th Gen Intel(R) Core(TM) i7-1185G7 @ 3.00GHz
USRP: Ettus Research USRP B210
driver: UHD 4.3.0.0-0ubuntu1~focal1
COTS UE: OnePlus 10 Pro 5G

## Expected Behavior ##
For the phone to be able to establish a connection at the AS and NAS layers

## Actual Behaviour ##
Phone cannot establish an RRC connection

## Steps to reproduce the problem ##
I run sudo srsenb and toggle airplane mode on the phone twice. (I also tried rebooting, same result)

## Additional Information ##
The phone does the see the network when scanning for available operators, upon trying to connect to it, I receive an error message stating that the network service is unavailable.

I tried previous suggestions such as decreasing the MCS to 10, tried different gain values, and dl_earfcn values as well as:
device_args = clock=external (although this did produce an error: Could not lock reference clock source. Sensor: ref_locked=false)

I am unsure of how to proceed from here. 

[enb.log](https://github.com/srsran/srsRAN/files/9761848/enb.log)
[enb.txt](https://github.com/srsran/srsRAN/files/9761850/enb.txt)

helloTkk, 2022-10-23:
Hi,

I'm also using Oneplus 10 Pro and encountered the same problem as you... Did you figure it out? Welcome to discuss with me, my email is: tukai2259@gmail.com.

helloTkk, 2022-10-23:
Hi,

I'm also using Oneplus 10 Pro and encountered the same problem as you... Did you figure it out? Welcome to discuss with me, my email is: tukai2259@gmail.com @Salous-99

andrepuschmann, 2022-11-07:
@Salous-99 Could you please share a eNB PCAP as well? Thanks