srsran/srsRAN_4G #1021: OnePlus 10 pro cannot finish the RRC connection

helloTkk, 2022-10-27:
## Issue Description ##
Dear srsRAN community, 

Hello! 

I followed this [note](https://docs.srsran.com/en/latest/app_notes/source/5g_sa_COTS/source/index.html), The srsenb and Open5GS works properly, and I have 2 COTS UE can successfully connect to it.

But for my OnePlus 10 pro, it can see the 5G SA cell and send several RACH messages, but cannot finish the RRC procedure as shown in the pcap, although I have decrease both of nr_pdsch_mcs and nr_pusch_mcs to 3 at enb.conf.

## Setup Details ##
Below is my setup:
Core network: Open5GS running on Ubuntu 20.04
gNB: srsenb
CPU: i7-12700k
USRP: Ettus Research USRP B210 with 2 VERT 900 antennas and a GPSDO
driver: UHD 3.15
COTS UE: OnePlus 10 Pro 5G

## Expected Behavior ##
OnePlus 10 pro connect to 5G SA network successfully.

## Actual Behaviour ##
UE can see the 5G cell but can’t connect to 5G SA network successfully, even cannot finish the RRC procedure. UE is not sending RRC complete messages.
[enb.txt](https://github.com/srsran/srsRAN/files/9875565/enb.txt)
[enb.log](https://github.com/srsran/srsRAN/files/9875566/enb.log)
[image]

## Additional Information ##
enodeB log and enodeB configuration file are attached. I’m really unsure of how to proceed from here, any help I’ll appreciate!

andrepuschmann, 2022-11-07:
Please send full logs in info and the PCAP please.

andrepuschmann, 2022-11-07:
It's likely the same issue as in https://github.com/srsran/srsRAN/issues/1010

helloTkk, 2022-11-08:
Hi @andrepuschmann,
Thanks for your reply. Here is the logs in info and PCAP. I appreciate your help!
[enb.log](https://github.com/srsran/srsRAN/files/9957159/enb.log)
[enb_mac_nr.zip](https://github.com/srsran/srsRAN/files/9957162/enb_mac_nr.zip)

andrepuschmann, 2022-11-08:
Looking at the logs again its clear that the UL signal quality is the issue. The SNR is very weak. You need to improve this by adjusting the gains for rx/tx on both sides, also make sure that you're not getting interference from other cells on that band. Check the logs of the eNB and observe the `snr=xx` value.

mode51software, 2023-01-30:
I have the same  issue using the OnePlus 10T with Android 12 and 13. 

I also have the issue when using the OnePlus Nord 8 on Android 12. With Android 11 the OnePlus Nord 8 works fine. The chipset is clearly physically the same but the problem with the RRC Setup happens when upgraded to Android 12. So I think the upgrade must also modify the baseband behaviour in some way and the 10 Pro, 10T and Nord 8 all exhibit the same problem with srsRAN.

I note the 10T and Nord 8 (Android 12) work ok with OAI on n78.

alexjiao2021, 2023-02-11:
Same here. Suspect some parameters in RRC setup message that UE doesn't like, any chance developers can take a look?

helloTkk, 2023-02-11:
Hi. May I know what SDR/usrp model you are currently using?

alexjiao2021, 2023-02-11:
It's LimeSDR.

mode51software, 2023-02-11:
I use the B205mini-i with an external GPSDO

tonyprotocols, 2023-02-16:
Hi, guys  I am dealing with the exact same problem. I think the problem is happening because gNB never allocate uplink resource block and never sends PUCCH dci=0_1 format message. I am not sure why becouse in the log file I can't see any error, if someone already fix this please share. Thank you!

alexjiao2021, 2023-02-17:
Should be because of ue-specific DCI format 0-0 and 1-0 in RRC setup message, should be 0-1 and 1-1.
Could you please take a look? @andrepuschmann 
[image]

andrepuschmann, 2023-02-17:
Hey folks,
sorry for the delay answering here. As you might know we are working on a completely new gNB for quite some time now. We will release a first version of it next week so I would ask you guys to try this again with the new release.
Thanks

alexjiao2021, 2023-02-18:
> Hey folks, sorry for the delay answering here. As you might know we are working on a completely new gNB for quite some time now. We will release a first version of it next week so I would ask you guys to try this again with the new release. Thanks

Hi Andre, will the new gNB support LimeSDR?

mode51software, 2023-03-22:
Hi I've tried with the new srsRAN O-RAN edition and the OnePlus 10T progresses through the RRC procedure now. 

I've tried TDD n78 and FDD n1, n3 and n7.

I get a registration reject but that appears to be caused by a separate issue with network slicing conf.

andrepuschmann, 2023-03-22:
Hey all, @helloTkk . can you please also use the new repo instead of this one. We won't be backporting any fixes to the SA gNB in this repo. I am closing this issue here - please create an issue in the new repo should you still see issues.

closed 2023-03-22 by andrepuschmann

mode51software, 2023-03-22:
Hi @helloTkk it is a B205mini with an external GPSDO enabled in the conf with clock: external. 

I used 20MHz bandwidth and 15khz scs for n1, n3 and n7, derived from the n78 example conf.