srsran/srsRAN_4G #1023: srsRAN with ZMQ Virtual Radios ( compile srsRAN failed)

wngandonang, 2022-10-29:
## Issue Description ##

I tried to compile srsRAN, but the make got error. Need help please. 

## Setup Details ##
I tried to build srsRAN, I respected all the steps which is described here ( https://docs.srsran.com/en/latest/app_notes/source/zeromq/source/index.html#zeromq-installation) .
Of course , I read this page ( https://docs.srsran.com/en/latest/general/source/1_installation.html) talking about the librairies that I needed to install.

Despite that, the make failed at 81% , I don't know , how  I can solve this problem. I tried to upgrade the cmake version ( I took the last 3.24.2 ) and the compiler gcc 8.4.0, but I got the same error. 

## Expected Behavior ##
The error talk about buffer , I don't know if any one, got the same thing ?  

## Actual Behaviour ##
The error message :
/home/nang/srsRAN/srsue/src/stack/upper/test/nas_test.cc:37:47: error: static assertion failed: liblte buffer and byte buffer members misaligned
 static_assert(alignof(LIBLTE_BYTE_MSG_STRUCT) == alignof(byte_buffer_t),
               ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~
srsue/src/stack/upper/test/CMakeFiles/nas_test.dir/build.make:75: recipe for target 'srsue/src/stack/upper/test/CMakeFiles/nas_test.dir/nas_test.cc.o' failed
make[2]: *** [srsue/src/stack/upper/test/CMakeFiles/nas_test.dir/nas_test.cc.o] Error 1
CMakeFiles/Makefile2:11647: recipe for target 'srsue/src/stack/upper/test/CMakeFiles/nas_test.dir/all' failed
make[1]: *** [srsue/src/stack/upper/test/CMakeFiles/nas_test.dir/all] Error 2
Makefile:165: recipe for target 'all' failed
make: *** [all] Error 2
 

## Steps to reproduce the problem ##
I worked on Ubuntu 18.04.06

andrepuschmann, 2022-11-07:
We built for Ubuntu 18.04 in the CI with gcc 7.5 without any issues. Not sure if you mixed something up on your installation. Please try a fresh build on a vanilla installation. If the issue still persists, please send full build logs with cmake, etc.

leohuangyy, 2024-08-26:
I met the same problem too,and i use with ubuntu16.04 32bit

nas_test.cc:37:1: error: static assertion failed: liblte buffer and byte buffer members misaligned
 static_assert(alignof(LIBLTE_BYTE_MSG_STRUCT) == alignof(byte_buffer_t),
 ^
srsue/src/stack/upper/test/CMakeFiles/nas_test.dir/build.make:62: recipe for target 'srsue/src/stack/upper/test/CMakeFiles/nas_test.dir/nas_test.cc.o' failed
make[2]: *** [srsue/src/stack/upper/test/CMakeFiles/nas_test.dir/nas_test.cc.o] Error 1
CMakeFiles/Makefile2:14602: recipe for target 'srsue/src/stack/upper/test/CMakeFiles/nas_test.dir/all' failed
make[1]: *** [srsue/src/stack/upper/test/CMakeFiles/nas_test.dir/all] Error 2
Makefile:160: recipe for target 'all' failed
make: *** [all] Error 2
root@openbts-virtual-machine:/home/openbts/srsLTE/build#