srsran/srsRAN_4G #1027: Follow up of issue #990

helloTkk, 2022-11-08:
Dear srsRAN community,

Hello. This is a follow up of the issue #990 opened in Github(https://github.com/srsran/srsRAN/issues/990).

My setup is shown in the issue description but this time I added a GPSDO.

However, after using GPSDO, the issue remains the same... I am sure I followed this [note](https://docs.srsran.com/en/latest/app_notes/source/5g_sa_COTS/source/index.html) and use this command successfully (./pySim-shell.py -p0 --script ./scripts/deactivate-5g.script) to configure the SIM card. After that, my Huawei P40 pro and Google Pixel 6 works properly with srsRAN 5G SA cell.

I have tried band 3 and band 7, and change MMC, MNC code as 00101, 00102, 90170, and even a commercial PLMN, none of them are working for my Oneplus Nord 5G AC2003... it still cannot see the 5G SA network...

I also tried OpenairInterface 5G SA, which are using band 78, this phone can successfully search it. 

Since the OnePlus Nord 5G AC2003 is the tested device in [official document](https://docs.srsran.com/en/latest/app_notes/source/5g_sa_COTS/source/index.html) so I think it should be able to connect with srsRAN 5G SA cell, could anyone give me some help in this case? 

By the way, the android version for my Oneplus Nord 5G AC2003 is android 11, the phone has already rooted. Baseband version is: .c9-00023-SAIPAN_GEN_PACK-1.422164.2.446371.2 

If anyone could use Oneplus Nord 5G AC2003 to connect with srsRAN 5G SA cell, could you please let me know the android version and baseband version of your Oneplus Nord 5G AC2003? I'd also like to see your enb.conf and which usrp you are using. Thanks in advance!

I would appreciate it if anyone could help me out!

helloTkk, 2022-11-09:
Hi @gabri94,
 
I'm using Sysmocom USIM(sysmoUSIM-SJA2), IMSI start with 00101, I'm sure that is matched with my gnb settings.
 
May I know your Oneplus Nord 5G AC2003 android version and baseband version?
 
Thanks.

fllay, 2022-11-17:
I have the same problem. did you try Network Signal Guru as suggested in the note?

dmielnik, 2022-12-03:
@helloTkk did you manage to overcome this problem ? I have the same issue with OnePlus Nord CE 2 Lite 5G (using srsRAN B200mini, GSPDO, sysmoUSIM-SJA2). I can not see the 5G SA network, LTE works fine though.

helloTkk, 2023-01-13:
> @helloTkk did you manage to overcome this problem ? I have the same issue with OnePlus Nord CE 2 Lite 5G (using srsRAN B200mini, GSPDO, sysmoUSIM-SJA2). I can not see the 5G SA network, LTE works fine though.

Hi, unfortunately, I still facing this issue. Did you overcome this problem?

mode51software, 2023-01-18:
I paid for 1 month of Network Signal Guru usage and found that by default OnePlus Nord 8 (AC2003) only looks for n78. Once I enabled the other bands, then it can see and attempt to register on srsRAN (though I have a separate issue to do with network slicing conf). In the screenshot "Preferred NR5G Band" should really be "Enabled NR5G Bands".

I've looked at the inbuilt engineering screens and not seen an option to change this (Qualcomm chipset) setting elsewhere. 

I note I've also modified the carrier_policy.xml in the baseband so will check if this is needed as well. I found that an unassigned 234- PLMN worked ok whereas 001-01 doesn't have 5G enabled. Perhaps changing the band support using NSG overrides this though:

[image]