srsran/srsRAN_4G #1038: srsRAN with Epiq Sidekiq

karkiekie, 2022-11-18:
## Issue Description ##
I have everything I need from Epiq built and installed. I have SoapySDR and SoapySDRSidekiq (the Fork from Epiq - NOT the one from Pothosware) installed and built. Now what we're getting on the make of srsRANv21.10, is an error in the ue_scanner.cc file at the #include <uhd.h>. Epiq Sidekiq code has no uhd.h in it and does not require it.

If I remove it the #include <uhd.h> from ue_scanner.cc, I get all kinds of "template with C Linkage" errors.
i.e:

In file included from /usr/include/c++/9/string:48,
                 from /usr/include/c++/9/bits/locale_classes.h:40,
                 from /usr/include/c++/9/bits/ios_base.h:41,
                 from /usr/include/c++/9/ios:42,
                 from /usr/include/c++/9/istream:38,
                 from /usr/include/c++/9/sstream:38,
                 from /usr/include/c++/9/complex:45,
                 from /home/nyquist/srsRanSuite/srsRANv21.10/lib/include/srsran/srsran.h:26,
                 from /home/nyquist/srsRanSuite/srsRANv21.10/lib/include/srsran/phy/rf/rf_utils.h:30,
                 from /home/nyquist/srsRanSuite/srsRANv21.10/lib/examples/ue_scanner.cc:40:
/usr/include/c++/9/bits/stl_function.h:1021:3: error: template with C linkage
 1021 |   template<typename _Predicate>
      |   ^~~~~~~~
In file included from /home/nyquist/srsRanSuite/srsRANv21.10/lib/examples/ue_scanner.cc:40:
/home/nyquist/srsRanSuite/srsRANv21.10/lib/include/srsran/phy/rf/rf_utils.h:26:1: note: ‘extern "C"’ linkage started here
   26 | extern  "C" {
      | ^~~~~~~~~~~
In file included from /usr/include/c++/9/string:48,
                 from /usr/include/c++/9/bits/locale_classes.h:40,
                 from /usr/include/c++/9/bits/ios_base.h:41,
                 from /usr/include/c++/9/ios:42,
                 from /usr/include/c++/9/istream:38,
                 from /usr/include/c++/9/sstream:38,
                 from /usr/include/c++/9/complex:45,
                 from /home/nyquist/srsRanSuite/srsRANv21.10/lib/include/srsran/srsran.h:26,
                 from /home/nyquist/srsRanSuite/srsRANv21.10/lib/include/srsran/phy/rf/rf_utils.h:30,
                 from /home/nyquist/srsRanSuite/srsRANv21.10/lib/examples/ue_scanner.cc:40:
/usr/include/c++/9/bits/stl_function.h:1028:3: error: template with C linkage
 1028 |   template<typename _Predicate>

karkiekie, 2022-11-18:
Update: I commented out the following lines in: /home/nyquist/srsRanSuite/srsRANv21.10/lib/include/srsran/phy/rf/rf_utils.h:26:1: 

//#ifdef __cplusplus
//extern  "C" {
//#endif

#include "srsran/phy/rf/rf.h"
#include "srsran/srsran.h"

typedef struct SRSRAN_API {
  uint32_t max_frames_pbch;      // timeout in number of 5ms frames for MIB decoding
  uint32_t max_frames_pss;       // timeout in number of 5ms frames for synchronization
  uint32_t nof_valid_pss_frames; // number of required synchronized frames
  float    init_agc;             // 0 or negative to disable AGC
  bool     force_tdd;
} cell_search_cfg_t;

SRSRAN_API int rf_rssi_scan(srsran_rf_t* rf, float* freqs, float* rssi, int nof_bands, double fs, int nsamp);

SRSRAN_API int
rf_mib_decoder(srsran_rf_t* rf, uint32_t nof_rf_channels, cell_search_cfg_t* config, srsran_cell_t* cell, float* cfo);

SRSRAN_API int rf_cell_search(srsran_rf_t*       rf,
                              uint32_t           nof_rf_channels,
                              cell_search_cfg_t* config,
                              int                force_N_id_2,
                              srsran_cell_t*     cell,
                              float*             cfo);

SRSRAN_API int rf_search_and_decode_mib(srsran_rf_t*       rf,
                                        uint32_t           nof_rf_channels,
                                        cell_search_cfg_t* config,
                                        int                force_N_id_2,
                                        srsran_cell_t*     cell,
                                        float*             cfo);

SRSRAN_API int rf_cell_search_nbiot(srsran_rf_t* rf, cell_search_cfg_t* config, srsran_nbiot_cell_t* cell, float* cfo);

//#ifdef __cplusplus
//}
//#endif

And now I get these errors:

[ 54%] Linking CXX executable ue_scanner
/usr/bin/ld: CMakeFiles/ue_scanner.dir/ue_scanner.cc.o: in function `main':
ue_scanner.cc:(.text.startup+0x5ed): undefined reference to `rf_search_and_decode_mib(srsran_rf_t*, unsigned int, cell_search_cfg_t*, int, srsran_cell_t*, float*)'
collect2: error: ld returned 1 exit status
make[2]: *** [lib/examples/CMakeFiles/ue_scanner.dir/build.make:101: lib/examples/ue_scanner] Error 1
make[1]: *** [CMakeFiles/Makefile2:8205: lib/examples/CMakeFiles/ue_scanner.dir/all] Error 2
make: *** [Makefile:163: all] Error 2

Please advise. I appreciate your help.

andrepuschmann, 2022-11-19:
It seems you're building a custom fork of srsRAN, we don't have a file called `ue_scanner.c`. I'd advice you build try the latest public release and report back. Also if you use the Soapy driver you don't need the native Epiq support - and vice versa.

karkiekie, 2022-11-19:
Hey there. Thank you for your response. We do indeed have our customized srsRan. It works great with a B210 with uhd drivers.
But now we have moved to the Sidekiq. We have the Sidekiq SDK, but when building srsRan, it doesn't recognize RF without the SoapySDR and SoapySDRSidekiq drivers. That is why I installed them, and that resolved that issue. I removed the line of code importing the uhd.h file, and I commented out the "//#ifdef __cplusplus" statements. But now I get this error that I am not sure how to resolve: 

Linking CXX executable ue_scanner
/usr/bin/ld: CMakeFiles/ue_scanner.dir/ue_scanner.cc.o: in function main': ue_scanner.cc:(.text.startup+0x5ed): undefined

I appreciate your guidance!

karkiekie, 2022-11-19:
Looking through the code now to localize the error.

andrepuschmann, 2022-11-21:
Fine, I am going to close the issue as it's not an issue in mainline srsRAN code.

closed 2022-11-21 by andrepuschmann