srsran/srsRAN_4G #1045: How to increase the tx power of enb or the coverage of enb?

coolcoollly, 2022-11-24:
## Issue Description ##
Hello!

I build a lte epc and enb with the tourial: https://docs.srsran.com/en/latest/app_notes/source/cots_ue/source/index.html#
I connect my phone with the enb successfully. But I find that the coverage of the enb is only about 30m, and I think  the reason is too low tx power of usrp antenna. I use my phone to measure the rsrp of the enb, and I find the rsrp range is from -75dbm to -125dbm. When I move phone from close to enb to 30m away from the enb, the rsrp move from -75dbm to -125dbm and disconnection happens. 
But I also measure the rsrp of commercial base stations, the values are almost -60dm or higher. So I think increasing the tx power of enb may improve the coverage of enb.
However, I try to set tx_gain in enb.conf from 0db to 100db, the measured rsrp doesn't change which is always  -75dbm to -125dbm.
And I know the max power of tx_port without antenna of usrpb210 is 12dbm from an agent. If so,  shouldn't the power near the enb be 15dbm(12dbm tx power+ 3dbi antenna gain). But the actual measured power is -75dbm, which I cannot understand.
Has somebody tried to change tx power of usrp, please give me some advice.

Thanks!   

## Setup Details ##
[Specify details of the test setup. This would help us reproduce the problem reliably]
e.g. Network configuration, Operation System, Hardware, RF front-end, library and driver versions
ubuntu 18.04; USRP2954; antenna VERT2450(3dbi gain); srsran22.04

## Expected Behavior ##
[What you expect to happen]

## Actual Behaviour ##
[What happens instead e.g. error message]

## Steps to reproduce the problem ##
[Tell us how to reproduce this issue e.g. RF setup, application config files]

## Additional Information ##
[Any additional information, configuration or data that might be necessary to reproduce the issue]

iman432, 2022-11-29:
Hi,
you should consider path loss, penetration loss and ... , because of these losses you see -75dbm instead of 15dbm and also commercial base stations have much more power from 30 watt to 120 watt per port but 12dbm is only 0.01 watt.

[image]

zikkuratti, 2022-12-02:
You need some overpriced Amp ~3 - 35W to get 1+km