srsran/srsRAN_4G #1046: Paging failed after RRC timer expired. ZMQ

scalessio, 2022-11-29:
## Issue Description ##
Hi, I am trying to run a basic setup of the 5G NR with ZMQ, 1UE and 1gNB. I managed to connect the UE,gNB and open5gs core.
After the attach procedure I can exchange traffic in both directions, UE<-->gNB and it is stable.

The issue is that after the RRC inactivity timer expires, the gNB is not able to ping the UE anymore. The core log output:
**[gmm] WARNING: [imsi-901700123456789] Paging failed. Stop (../src/amf/gmm-sm.c:296)**
In the meantime, the UE can still ping the AMF, even the last message from the UE is _Received RRC Release_.

I have also tried to reduce/increase the **rrc_inactive_timer** in the enb.conf to isolate the problem.
To me, It looks like the gNB is not able to handle the state transition between INACTIVE <---> CONNECTED. And it is not clear to me if upon receiving the RRC release message the UE pass to the INACTIVE or in IDLE as in the LTE.

Does someone had noted the same problem?

## Setup Details ##
Run on Alienware-13-R3 with Ubuntu 18.04. Install the core with open5gs and use srsRAN with zmq. 
I use the last release of srsRAN 20.10

## Expected Behavior ##
I expect that on the UE a message "RRC resume request" appears after the paging message from the core. 

## Actual Behaviour ##
1- Nothing from the UE. The last message is _Received RRC Release_
2-The gNB can not ping the UE.
3-In the core

## Steps to reproduce the problem ##
I followed the tutorial in the official srsRAN website, I just use the band 20 in the NR.
rr.conf:
    rf_port = 0;
    cell_id = 1;
    root_seq_idx = 1;
    tac = 7;
    pci = 500;
    dl_arfcn = 160200;
    //ul_arfcn = 514000;
    band = 20;

sudo srsenb
sudo srsue --rat.nr.dl_nr_arfcn 160200 --rat.nr.ssb_nr_arfcn 160350

Attached the log of the enb.
[enb_log.txt](https://github.com/srsran/srsRAN/files/10117606/enb_log.txt)

kaposnick, 2022-12-14:
I have faced the same issue and it seems indeed is the same problem. For my experiments, I have increased the inactivity timer to as much as the experiment lasts in order to avoid this issue.

scalessio, 2022-12-15:
Thanks for your reply N. Yes, I also figured out that in the 5G-SA the rrc resume procedure is not implemented yet (or completely implement). This is the main issue.  When the rrc timer expires and UE goes into the inactive state, you need to have the RRC Resume procedure to go back to the connected state.

This is also true when UE detects radio link failure or lost sync. They're not able to do the connection rrc reestablishment yet.

pksahoo2022, 2023-01-02:
@scalessio @kaposnick 
Hi i am also facing the same issue. Can you guys help me to resolve this if you have done it already ?
Attaching the outputs for reference.
[image]
[image]

andrepuschmann, 2023-01-09:
It's a known limitation of this version. I am afraid we won't provide a fix for it either. Please wait for the new srsGNB release (coming in Q1/23) which will have support for it.

closed 2023-01-09 by andrepuschmann

scalessio, 2023-01-09:
Thanks for your reply, Andre.

Do you know if the new srsGNB release will also implement/fix the problem of RRC reconnection request?

The current version doesn’t implement the reconnection properly afaik.

Regards.

-Alessio

andrepuschmann, 2023-01-09:
Yes, also this will be working