srsran/srsRAN_4G #1047: Paging issue with ZMQ, GnuRadio and 2 UEs

nedsana, 2022-12-02:
Hi, 
I am trying to create simple telecom system consisting of EPC, 1xENB and 2xUE. I'd like to use ZMQ and Gnuradio to experiment with the radio channel at some point. 
Based on the following discussion (https://github.com/srsran/srsRAN/issues/818) I was able to achieve my setup. Radio data now flows between the ENB and the UEs.

But something is not right with the UE behavior. Based on stdout, I can see the following:
- UE1 goes to "RRC Connected" state, obtains an IP, but seems to stuck there and never goes to "RRC IDLE" state.
- UE2 goes to "RRC Connected" state, but then "Radio-Link Failure" is reported and attach seems to fail. 
  Some time later a new attempt is made, which is successfull and IP is obtained.

At this point of time, we can ping UE2, but not UE1. And that behavior is repeatable, just sometimes the UEs may change roles (UE1 will be ping-able and UE2 not).

For the description above, I used srsRAN version release_22_10. 

I decided to try the same setup (with ZMQ) with older srsRAN - release_20_10_1. And there all seems fine. 
Both UEs will connect, both will obtain IP, both can go to "RRC IDLE" state, both can be pinged.

I think there is some regression here. Could you, please comment on this? Thank you!

**Configuration:**
Ubuntu 20.04, CPU:i7-9850H, 6x2.60GHz (12 threads), RAM:16GB
- srsRAN version to cause issue: release_22_10, 254cc719a9a31f64ce0262f4ca6ab72b1803477d
- srsRAN version to compare against: release_20_10_1, 45486b6e2c622ab1d47906521662f087816d7d03

**Steps to recreate the issue (same for release_22_10 and release_20_10_1):**
1. Create network space for ue1: sudo ip netns add ue1
2. Create network space for ue2: sudo ip netns add ue2
3. Verify creation: sudo ip netns list
4. Start EPC: sudo ./srsepc/src/srsepc ~/.config/srsran/epc.conf
5. Start GnuRadio flowgraph
6. Wait 10sec (may be this step is not necessary)
7. Start ENB: ./srsenb/src/srsenb ~/.config/srsran/enb.conf --rf.device_name=zmq --rf.device_args="fail_on_disconnect=true,tx_port=tcp://127.0.0.1:2101,rx_port=tcp://127.0.0.1:2100,id=enb,base_srate=23.04e6"
8. Start UE1: sudo ./srsue/src/srsue ~/.config/srsran/ue1.conf --rf.device_name=zmq --rf.device_args="tx_port=tcp://127.0.0.1:2001,rx_port=tcp://127.0.0.1:2000,id=ue1,base_srate=23.04e6" --gw.netns=ue1
9. Start UE2: sudo ./srsue/src/srsue ~/.config/srsran/ue2.conf --rf.device_name=zmq --rf.device_args="tx_port=tcp://127.0.0.1:2011,rx_port=tcp://127.0.0.1:2010,id=ue2,base_srate=23.04e6" --gw.netns=ue2
10. Wait untill both UEs go to "RRC IDLE" state (~30sec). In the problematic case (rel_22_10), one UE, will never get to this state.
10. Generate DL traffic: ping 172.16.0.3 -c3 -i3 -> OK
11. Generate DL traffic: ping 172.16.0.2 -c3 -i3 -> NOTE: IP 172.16.0.2 is unreachable in my setup. This is the issue.

Another observation: if I run EPC and ENB from rel_20_10_1 and UEs from rel_22_10, the UEs, behave correctly.

I am attaching:
1. srsRAN_Logs.rel_22_10.zip - contains srsRAN config files, PCAPs, stdout logs and log files for epc, enb, and both ues (the problematic behavior) [srsRAN_Logs.rel_22_10.zip](https://github.com/srsran/srsRAN/files/10141495/srsRAN_Logs.rel_22_10.zip)

2. srsRAN_Logs.rel_20_10_1.zip - contains srsRAN config files, PCAPs, stdout logs and log files for epc, enb, and both ues (the correct behavior) [srsRAN_Logs.rel_20_10_1.zip](https://github.com/srsran/srsRAN/files/10141496/srsRAN_Logs.rel_20_10_1.zip)

Thank you!

nedsana, 2022-12-06:
Hi kaposnik, 

Thanks for the response. Indeed, the context file seems to be the problem. Commenting the lines, you mention, and both ues were able to connect. 

I inspected a bit further, though, and I think the problem is due to the overwriting of the file. We have 2 UEs, both trying to store and read UE specific data in one and the same file. Obviously the context file will contain the data of the last UE. Then on startup, both UEs get this same data. One can connect. But the other will try to connect with the same settings, and I guess, ENB or EPC will discard it.

My solution was to extend the configuration file for the NAS section. I added "ctxt_file" parameter. Now both UEs read and write their own file. No collision and both can connect now.

Closing the issue.

closed 2022-12-06 by nedsana

Jeffolobo123, 2023-07-02:
Hola q tal, buenas noches con todos: estoy trabajando con una estación radio base, instalé ya todos los programas para un UE, necesito saber como puedo configurar para más usuarios si abría la posibilidad sin tener q usar ZMQ, ya que no me reconoce otro sim card. gracias

wmwwimor, 2023-08-02:
> Hi kaposnik,
>
> Thanks for the response. Indeed, the context file seems to be the problem. Commenting the lines, you mention, and both ues were able to connect.
>
> I inspected a bit further, though, and [...]

Could you tell me how to revise ? I found that when two ues  start to  connect to the net, they will get same prach_occasion,ra-rnti ,  c-rnti,ta,  and only one of them can get an IP from 5gc
I've been stuck for two days.
Thanks!!