srsran/srsRAN_4G #1053: Problem with SRSRAN and the timestamp solution

LunaWuna, 2022-12-07:
## Issue Description ##
Starting up a eNB on a plutosdr (the "pluto plus" "enhanced" version) brings up ``Error reading rx ringbuffer. Invalid header``

## Setup Details ##
Linux mint with a "pluto plus" SDR running the timestamp solution, the TXRX test works.

## Expected Behavior ##
eNB to start and a signal to be transmitted

## Actual Behaviour ##
[image]
This happens on both a linux mint VM and another PC running ubuntu on bare metal.

ofontbach, 2022-12-16:
A fix has been pushed in the timestamp repository. Please, update your local one and try again. Also, in case it is necessary, open a new issue in the timestamp repository (https://github.com/srsran/zynq_timestamping) not here. Thanks.

andrepuschmann, 2022-12-16:
Closing this issue as fixed. As mentioned by @ofontbach please use the timestamping repo in case there is anything else with it.

closed 2022-12-16 by andrepuschmann