srsran/srsRAN_4G #1054: Trying to use b71 with COTS UE

lhmscpqd, 2022-12-07:
Hello, community.

## Issue Description ##
Aiming to use b71 LTE, I've begun to make some changes in the srsRAN stack from tag "release_22_10" (more specifically changing one line - b71 with correct values - and adding another in the struct lte_bands - adding b72). After some time, I discovered that an open pull request (#973) is making the exactly same implementation  I've done.

## Setup Details ##
- srsRAN tag: release_22_10
- OS: Ubuntu 18.04 desktop (physical machine)
- RAM: 8 GB
- Disk: 512 GB
- CPU: 
```bash
Architecture:        x86_64
CPU op-mode(s):      32-bit, 64-bit
Byte Order:          Little Endian
CPU(s):              4
On-line CPU(s) list: 0-3
Thread(s) per core:  2
Core(s) per socket:  2
Socket(s):           1
NUMA node(s):        1
Vendor ID:           GenuineIntel
CPU family:          6
Model:               69
Model name:          Intel(R) Core(TM) i7-4600U CPU @ 2.10GHz
Stepping:            1
CPU MHz:             2992.959
CPU max MHz:         2701,0000
CPU min MHz:         756,0000
BogoMIPS:            5387.32
Virtualization:      VT-x
L1d cache:           32K
L1i cache:           32K
L2 cache:            256K
L3 cache:            4096K
NUMA node0 CPU(s):   0-3
Flags:               fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm cpuid_fault epb invpcid_single pti ssbd ibrs ibpb stibp tpr_shadow vnmi flexpriority ept vpid ept_ad fsgsbase tsc_adjust bmi1 avx2 smep bmi2 erms invpcid xsaveopt dtherm ida arat pln pts md_clear flush_l1d
```

## Expected Behavior ##
Be able to use b71 LTE in srsRAN with COTS UEs.

## Actual Behaviour ##
During my tests, both COTS UE Moto Edge Plus and Quectel RM500Q-GL failed to detect the RAN in b71 because SIBs weren't created correctly. The enb.log file points me to file .../srsRAN/lib/src/asn1/rrc/si.cc:3150, where the function "pack_integer" is called limiting the "ub" up to 64u.
I've tried to change this value to 72u, but then another error appeared in the log: in .../srsRAN/lib/src/asn1/rrc/si.cc:2613 the same "ub" limitation was observed, then, the same strategy was applied, setting a higher value (133471) to the field.
After these two modifications, no error or warning was appointed in enb's log, but the UEs don't detect the desired band. Collecting pcaps and configuring Wireshark as described in the documentation (https://docs.srsran.com/en/next/general/source/5_troubleshooting.html#examining-pcaps-with-wireshark), I noticed that Wireshark couldn't decode SIB2 messages when I used b71, as is showed bellow:
```bash
LTE RRC DL_SCH    SystemInformation [ SIB2[UNKNOWN PER: too long integer(per_normally_small_nonnegative_whole_number)]
```
This behavior did not occur in the vanilla version with bands lower than 64 (tested with bands 7, 20, and 28):
```
LTE RRC DL_SCH	SystemInformation [ SIB2 SIB3 ]
```
I would like to know if some additional configuration/change should be done to enable the usage of higher LTE bands (even in the vanilla version).

## Steps to reproduce the problem ##
Apply this patch to add the b71 definition in lte_bands:
```bash
From b4f6e9aaffb572b170f854a7d67d4b6ea96ec4eb Mon Sep 17 00:00:00 2001
From: Luiz H <lhsilva@cpqd.com.br>
Date: Tue, 6 Dec 2022 10:53:34 -0300
Subject: [PATCH] Add b71 LTE definition in lte_bands

---
 lib/include/srsran/phy/common/phy_common.h | 2 +-
 lib/src/phy/common/phy_common.c            | 3 ++-
 2 files changed, 3 insertions(+), 2 deletions(-)

diff --git a/lib/include/srsran/phy/common/phy_common.h b/lib/include/srsran/phy/common/phy_common.h
index f19b2ce87..79761a0b0 100644
--- a/lib/include/srsran/phy/common/phy_common.h
+++ b/lib/include/srsran/phy/common/phy_common.h
@@ -171,7 +171,7 @@ typedef enum { SRSRAN_SF_NORM = 0, SRSRAN_SF_MBSFN } srsran_sf_t;
 #define SRSRAN_FDD_NOF_HARQ (FDD_HARQ_DELAY_DL_MS + FDD_HARQ_DELAY_UL_MS)
 #define SRSRAN_MAX_HARQ_PROC 15
 
-#define SRSRAN_NOF_LTE_BANDS 58
+#define SRSRAN_NOF_LTE_BANDS 59 //58
 
 #define SRSRAN_DEFAULT_MAX_FRAMES_PBCH 500
 #define SRSRAN_DEFAULT_MAX_FRAMES_PSS 10
diff --git a/lib/src/phy/common/phy_common.c b/lib/src/phy/common/phy_common.c
index b4838ac53..be0f86de4 100644
--- a/lib/src/phy/common/phy_common.c
+++ b/lib/src/phy/common/phy_common.c
@@ -587,7 +587,8 @@ struct lte_band lte_bands[SRSRAN_NOF_LTE_BANDS] = {
     {68, 753, 67536, 132672, 55, SRSRAN_BAND_GEO_AREA_EMEA},
     {69, 2570, 67836, 0, 0, SRSRAN_BAND_GEO_AREA_EMEA},
     {70, 1995, 68336, 132972, 300, SRSRAN_BAND_GEO_AREA_NAR},
-    {71, 0, 68586, 133122, 0, SRSRAN_BAND_GEO_AREA_NAR} // dummy band to bound band 70 earfcn
+    {71, 617, 68586, 133122, -46, SRSRAN_BAND_GEO_AREA_NAR},
+    {72, 0, 68936, 133472, 0, SRSRAN_BAND_GEO_AREA_EMEA}
 };
 
 int srsran_str2mimotype(char* mimo_type_str, srsran_tx_scheme_t* type)
-- 
2.17.1

```

The following .conf files was changed:
```bash
# srsenb.conf

[enb]
enb_id = 0x01
mcc = 001
mnc = 01
mme_addr = 192.168.10.44
gtp_bind_addr = 192.168.10.135
s1c_bind_addr = 192.168.10.135
s1c_bind_port = 0
n_prb = 25

[enb_files]
sib_config = /home/oai4/luiz/sib.conf
rr_config  = /home/oai4/luiz/rr.conf
rb_config = /home/oai4/luiz/rb.conf

[rf]
tx_gain = 80
rx_gain = 40

[pcap]
enable = true
filename = /home/oai4/luiz/enb_mac.pcap
s1ap_enable = true
s1ap_filename = /home/oai4/luiz/enb_s1ap.pcap

[log]
all_level = debug
all_hex_limit = 32
filename = /home/oai4/luiz/enb.log
file_max_size = -1

[gui]
enable = false
...
```

```bash
# rr.conf
...
cell_list =
(
  {
    // rf_port = 0;
    cell_id = 0x01;
    //tac = 0x0007;
    tac = 0x0001;
    pci = 1;
    // root_seq_idx = 204;
    dl_earfcn = 66886;
    // B07 (FDD): 3350
    // B20 (FDD): 6340
    // B28 (FDD): 9310  [9210, 9659]
    // B39 (TDD): 38275 -> Não funciona
    // B71 (FDD): 68766 // 68761
    //ul_earfcn = 133147; ////ul_earfcn = 21400;
    ho_active = false;
    //meas_gap_period = 0; // 0 (inactive), 40 or 80
    //meas_gap_offset_subframe = [6, 12, 18, 24, 30];
...
```

To execute eNB, I used the following command:
```bash
sudo srsenb /home/oai4/luiz/enb.conf
```

Thanks in advance.

WVFmauricio, 2022-12-23:
### Issue Description
Being guide by the pull request #617 I have tried to make the b66 and b71 works. In my tests the SIB1 is working fine and sending the nonCriticalExtension containing the SystemInformationBlockType1-v8h0-IEs with the right freqBandIndicator-v9e0. 

### Actual Behaviour

However, I am facing a problem with SIB2 in the msg2. When including the SystemInformationBlockType2-v8h0-IEs, the msg2 got incomplete and some elements cannot be seem in the PCAP using wireshark, such as the SystemInformationBlockType2-v8h0-IEs and other SIBs (SIB3, SIB4 and so on). Follow an image of the problem.

[image]

### Steps to reproduce the problem
Apply this patch to add the b71 definition in lte_bands:

**/lib/include/srsran/phy/common/phy_common.h**
```  diff --git a/lib/include/srsran/phy/common/phy_common.h b/lib/include/srsran/phy/common/phy_common.h
index f19b2ce87..79761a0b0 100644
--- a/lib/include/srsran/phy/common/phy_common.h
+++ b/lib/include/srsran/phy/common/phy_common.h
@@ -171,7 +171,7 @@ typedef enum { SRSRAN_SF_NORM = 0, SRSRAN_SF_MBSFN } srsran_sf_t;
 #define SRSRAN_FDD_NOF_HARQ (FDD_HARQ_DELAY_DL_MS + FDD_HARQ_DELAY_UL_MS)
 #define SRSRAN_MAX_HARQ_PROC 15
 
-#define SRSRAN_NOF_LTE_BANDS 58
+#define SRSRAN_NOF_LTE_BANDS 59 //58
 
 #define SRSRAN_DEFAULT_MAX_FRAMES_PBCH 500
 #define SRSRAN_DEFAULT_MAX_FRAMES_PSS 10
  ```

**/lib/src/asn1/rrc/si.cc b/lib/src/asn1/rrc/si.cc**
  ```
diff --git a/lib/src/asn1/rrc/si.cc b/lib/src/asn1/rrc/si.cc
index 80b23294e..045b3e728 100644
--- a/lib/src/asn1/rrc/si.cc
+++ b/lib/src/asn1/rrc/si.cc
@@ -2653,6 +2653,9 @@ SRSASN_CODE sib_type2_s::pack(bit_ref& bref) const
       varlength_field_pack_guard varlen_scope(bref, false);
 
       HANDLE_CODE(bref.pack(late_non_crit_ext_present, 1));
+      if (late_non_crit_ext_present) {
+        HANDLE_CODE(late_non_crit_ext.pack(bref));
+      }
     }
     if (group_flags[1]) {
       varlength_field_pack_guard varlen_scope(bref, false);
@@ -2798,6 +2801,9 @@ SRSASN_CODE sib_type2_s::unpack(cbit_ref& bref)
       varlength_field_unpack_guard varlen_scope(bref, false);
 
       HANDLE_CODE(bref.unpack(late_non_crit_ext_present, 1));
+      if (late_non_crit_ext_present) {
+        HANDLE_CODE(late_non_crit_ext.unpack(bref));
+      }
     }
     if (group_flags[1]) {
       varlength_field_unpack_guard varlen_scope(bref, false);   
```
**/lib/src/phy/common/phy_common.c b/lib/src/phy/common/phy_common.c**
 ``` 
diff --git a/lib/src/phy/common/phy_common.c b/lib/src/phy/common/phy_common.c
index b4838ac53..be0f86de4 100644
--- a/lib/src/phy/common/phy_common.c
+++ b/lib/src/phy/common/phy_common.c
@@ -587,7 +587,8 @@ struct lte_band lte_bands[SRSRAN_NOF_LTE_BANDS] = {
     {68, 753, 67536, 132672, 55, SRSRAN_BAND_GEO_AREA_EMEA},
     {69, 2570, 67836, 0, 0, SRSRAN_BAND_GEO_AREA_EMEA},
     {70, 1995, 68336, 132972, 300, SRSRAN_BAND_GEO_AREA_NAR},
-    {71, 0, 68586, 133122, 0, SRSRAN_BAND_GEO_AREA_NAR} // dummy band to bound band 70 earfcn
+    {71, 617, 68586, 133122, -46, SRSRAN_BAND_GEO_AREA_NAR},
+    {72, 0, 68936, 133472, 0, SRSRAN_BAND_GEO_AREA_EMEA}
 };
 
[... 184 lines cut ...]
### Analysis of ENB logs Until now

I have analyzed the eng logs and in lines 58 to 60 there is the transmitted msg2 containing the SIB2 message. Follow the message 

```
2022-12-23T20:55:54.001362 [RRC    ] [I] Tx Broadcast PDU - systemInformation-r8, cc=0, idx=1 (34 B)
    0000: 00 02 0c e1 bf 78 88 00 ca 11 e0 01 00 00 08 01
    0010: 82 99 45 ab 9c 30 c7 ff fe 81 c0 41 60 6c 82 a3
```

The message "c0 41 60 6c 82 a3" should contain the LateNonCriticalExtension and the SystemInformationBlockType2-v8h0-IEs. Follow the message in binary

Message in hex -> c0 41 60 6c 82 a3
Message in Binary -> 11000000 01000001 01100000 01101100 10000010 10100011

In line 160 of enb logs, we can found the lateNonCriticalExtension message that sould have being sent. The values is  "lateNonCriticalExtension": "641518". Follow the message in binary

Message in hex -> 64 15 18
Message in Binary -> 01100100 00010101 00011000

However, the bits of lateNonCriticalExtension in line 160 cannot be found in the bits sent in line 58 to 60. This seems to be the problem. The message are not being sent correctly. 

**I would like to know if someone already have faced the same issue and how to fix it?  Did I missed to configure something? **

Thanks in advance!!

andrepuschmann, 2023-04-17:
Thanks for your detailed report @lhmscpqd and the update on it @WVFmauricio . PR #973 will be included in the next version but we didn't extend/add the changes to the ASN1 packing you suggested. So the issue will still be present. We also have no means to really test b71 I am afraid.