srsran/srsRAN_4G #1055: BladeRF srsenb Error - buffer pool is empty

s21sm, 2022-12-08:
## Issue Description ##
After running this: sudo srsepc
When I run this command: sudo srsenb  after a while it gives this message "00000000000000000Error - buffer pool is empty" and the same thing continues

## Setup Details ##
RF device = BladeRF
OS = DRAGON OS FOCAL ( as virtual machine with USB 3.0)

RamzuYunus, 2022-12-12:
I Hi. I get the same error. Have you found out the solution?

s21sm, 2022-12-13:
In  DRAGON OS FOCAL virtual machine I faced the mentioned problem. Using Ubuntu 20.04 I did not face this problem, however, I do not see any network name in my phone. I matched transmission frequency with my mobile's supported frequency but no result.

RamzuYunus, 2022-12-13:
> In DRAGON OS FOCAL virtual machine I faced the mentioned problem. Using Ubuntu 20.04 I did not face this problem, however, I do not see any network name in my phone. I matched transmission frequency with my mobile's supported frequency but no result.

Thanks. I was using Dragon OS as well. Now seems to work.