srsran/srsRAN_4G #1059: COTS UE can connect only to band 5

cyberkul, 2022-12-15:
## Issue Description ##
[Can not connect a Quectel EG-25G COTS UE]

## Setup Details ##
[SRSRAN v 22.4.1, UBUNTU 22.04LTS on Ryzen 7 PC with USB 3.0, USRP B-210. UE is connected with USB to Windows 10 PC running Qnavigator, which is used to send AT-commands to UE. Have tried to use 10MHz external reference on B210, and adding device_args = clock=external in enb.conf. Don`t know if this is correct?]

## Expected Behavior ##
[Successfully connect at different LTE-bands, I want to use this to test co-location behaviour with other transmitting equipment]

## Actual Behaviour ##
[Can sometimes connect if using dl_earfcn = 2550]

## Steps to reproduce the problem ##
[Don`t know]

## Additional Information ##
[I notice that there is a dl_earfcn setting in rr.conf, do I need to change this as well as in the enb.conf?]

cyberkul, 2022-12-16:
I move this to the mailing list, sorry..

tiger762, 2022-12-22:
I am dealing with much the same issue at the moment. I have an LG K7 (2018 vintage) that works, whereas IPhone 12 and OnePlus Model 5 do not discover the network. I have an Ettus B200. Also a 10MHz and 1PPS GPSDO I got off of a Chinese seller on Ebay. Verified the 10MHz output on my Tektronix TDS2012. It's a 1 volt amplitude sine wave. Into a 50 ohm load that is 10mw or 10dbm. Probably would be better if it were a square wave, because the point where the B200 perceives a logic-1 might not be precisely defined.

One theory I am exploring is something in the SIB is making the handset perceive this isn't a bonafide LTE network. Going to run 'srsue' on a real carrier to capture their SIBs, then compare... 

Good luck!

plaiddshirt, 2024-01-30:
Hi, can the module EC25G from Quectel that you are using connect the pc to another USRP equipped base station, I am also making an attempt to access the srs_RAN base station through the Quectel module, but the EC20 I have does not support this connection. Thanks a lot!

cyberkul, 2024-01-30:
Hi!
I have not come any further with this, except that someone suggested that
the SIM-card can be the problem. I have two, I`ll try to get new SIM-cards
and try again. The SIM-card should also support 5G.
Gunnar

plaiddshirt, 2024-01-30:
> Hi! I have not come any further with this, except that someone suggested that the SIM-card can be the problem. I have two, I`ll try to get new SIM-cards and try again. The SIM-card should also support 5G. Gunnar tir. 30. jan. 2024 kl. 15:36 skrev plaiddshirt ***@***.***>:
> […](#)

Hi!
Thanks for the reply!
I'm now using a USIM card with Quectel to try to access the pc to the srsRAN base station, which also keeps failing with the existing EC20 module.
And the EC200A-CN I'm using doesn't support access to base stations outside the list of carriers it is limited to.
So if you have any new progress, I also hope you can share it hahaha.
Thanks a lot!

cyberkul, 2024-01-30:
I'll let you know if I get any progress. Could be something simple, I don't know much about 4G. With srsUE on a different pc, and another Ettus B210 it works. Very slow to connect, but when connected it is stable. Have also tried different bands with success.Sendt fra min iPhone30. jan.